bigtreetech/BIGTREETECH-TFT35-V3.0 #79: Is there a User's guide describing the menu functions on the touchscreen

chobson303, 2020-09-02:
Is there a user's guide or document that describes what each function button on the TFT-V3.0 touchscreen?

oldman4U, 2020-09-02:
Hello.

As far as i know, there is no manual which describes all the functions of the touchscreen. Is there any particular function you would like to get explained or is this more a general question?

chobson303, 2020-09-03:
What are Babysteps and how do they work?

AnHardt, 2020-09-03:
Usually a move, in Marlin; is a 'segment'. Segments have a minimal length to not overload the planner. Babysteps don't have that limitation. Theoretically therewith single microsteps can be made. For practical cases that would cause a nearly infinite number of encoder_steps/key or screen presses. Therefore usually multiple microsteps (expressed as a fraction of a mm) are done at once. However, babysteps should allow to move a smaller then the normal minimal distance.

oldman4U, 2020-09-03:
In contrast to the character based screens which work directly with Marlin code on the mainboard, using the Touchscreen works a bit different. Imagine the TFT is "computer" by itself, sending strings with gCode to the mainboard, like Octoprint or Pronterface. But finally both methods are based on gCode, so i would always search for the corresponding code first here: https://marlinfw.org/docs/gcode/M290.html

I hope this helps further, but if not - Google and YouTube are your best friends when it comes to 3D printing - besides this GitHub and AnHardt of course;-)

Please do not forget to close this ticket, as soon as you do not need it anymore.

Thank you

closed 2020-09-03 by chobson303