bigtreetech/BIGTREETECH-TFT35-V3.0 #80: errors loading bed leveling mesh and saving to eeprom

nebakanezzer, 2020-09-05:
skr1.4t/2209/cr-10 S5

marlin bugfix 2.0

I get 'unable to load bed leveling' before every print, despite probing the bed before hand, saving it to a slot in the gui, then saving the eeprom, and having a load mesh portion in my gcode from my slicer (which worked before upgrading to the skr/tft as well as with other printers).

I would also get unable to load bed leveling upon turning on the printer, again, presumably because it thinks there is no mesh. I disabled this on the tft with auto_load_leveling:0  since it is already in the marlin firmware, and I no longer get this error, but it seems kind of odd to not match the firmware of the board

in fact, it seems like most issues with this display stem from the fact it is trying to half be a screen, half a board. a very weird thing that doesn't seem to have anywhere to adjust in the TFT but only happens with it, is when probing a mesh, it probes the entire bed, twice. I have marlin set to do a fast then slow probe with a 3 point grid. so I should have 9 points, probed twice. the printer does this....then starts all over, probing all 9 points, 2 more times. it is now taking ages to do an ABL and test the ABL saving errors.

there is also two places to save to eeprom, and I am not sure which actually does anything, and the first freezes the entire printer every now and then. these menus are machine->paramter->(scroll all the way down)->save settings and machine->custom->(scroll down)->save eeprom

oldman4U, 2020-10-09:
One month no answer. What is the status of the reported issue?

nebakanezzer, 2020-10-11:
> One month no answer. What is the status of the reported issue?

so not a fix, but a workaround I guess.

I realized that if I let the printer do its weird double probing of the bed...and let it sit for 10 seconds or so, I could then go to the save menu and save to eeprom. I then saved the mesh to slot0. the probe would then do a repeatability test... even though I didn't select that, which means it will probe one spot about 10-15 times. I was super curious... so again, I just let it do this and waited 10 seconds after it was complete. 

after this, the mesh loads every print perfectly without error.

oldman4U, 2020-10-12:
Have you tried latest Marlin version. I never use those premade configurations. They are usually very old and do not have the latest fixes.

nebakanezzer, 2020-10-12:
> Have you tried latest Marlin version. I never use those premade configurations. They are usually very old and do not have the latest fixes.

It's not premade, that is my own fork of marlin I built from the most recent bigfix branch, hence why my user name is in the link

oldman4U, 2020-10-13:
Ok. There are some fixes and new BL related functions on the way. Check the list of PRs waiting to be released in the tft firmware section.

oldman4U, 2020-10-27:
Have you tried the latest TFT firmware with the Mesh Editing stuff and all the fixes related to software end stop?