bigtreetech/BIGTREETECH-TFT35-V3.0 #81: TFT35 E3 V3.0 only shows the text 'LCD12864 simulator' nothing else even after turning knob

tony1ant, 2020-09-06:
Hi Every1

I've installed a BTT TFT35 E3 V3.0 on my Creality Ender 3 Pro & the 'Touch Mode' works fine. However in 'Marlin Mode' it only shows the text 'LCD12864 Simulator' on the screen & nothing else, even after turning the dial knob. I've DLoaded the latest Firmware from Marlin & from BigTreeTech (BTT). I've flashed the firmware numerous times with just the basics to get me going and also with all my Mods (below) enabled and I still get the same result every time: & it only shows the text 'LCD12864 Simulator' on the screen.
I've followed the instructions to the letter from BTT's Instructions and from 3 different Youtube tutorials (Teaching Tech, Chris Riley & others) and I still get the same result, No Marlin screen. All 3 cables for the screen R connected correctly & I have used brand new cables and parts for everything.
I've been trying for weeks now and now I've got No Hair left :(( . 
I'll B Eternally Grateful if some1 could help me with this and get my screen working in both modes. 

I hava Creality Ender 3 Pro with the following Mods:
BTT SKR E3 V1.4Turbo—TMC22209's
BTT TFT35 E3 V3.0 Touchscreen
BL-Touch
Direct Drive (STD CE3 Hotend)
BTT UPS
BTT Power Off relay V1.2   (tried enabled but then I disabled until I get the basics working)
DC to DC module
Wi-Fi Module ESP-01S   (tried enabled but then I disabled until I get the basics working)
Filament Runout Sensor  
NeoPixels 12 LED ring   (tried enabled but then I disabled until I get the basics working)
Raspberry Pi 3B+ with OctoPrint   (tried enabled but then I disabled until I get the basics working, Pi 3B+ was working perfectly B4 the BTT New Board & Mods)

I've enabled the following for the TFT35 E3 V3.0 touchscreen:
REPRAP_DISCOUNT_FULL_GRAPHIC_SMART_CONTROLLER in configuration.h
&
CR10_STOCKDISPLAY in configuration.h

ThankYou in advance, & Stay Safe

BobP2020, 2020-09-06:
Try commenting out the "CR10_STOCKDISPLAY" as you are not using it.

tony1ant, 2020-09-06:
#BobP2020 I shall try your suggestion. 
Thank U 4 the Super Fast reply :)

oldman4U, 2020-09-07:
Hi.

Your description is a bit confusing: BTT SKR E3 V1.4Turbo—TMC22209's

The mainboard is finally a SKR v1.4 Turbo which does not include E3 in its name. This is important, because all E3 boards are using a single EXP connector to connect the TFT with the mainboard and CR-10 Stockdisplay has to be activated in firmware. All NON E3 boards are connected using 2 EXP connectors and the RepRap display has to be activated in firmware.

Please do not forget to close this ticket, once you do not need it anymore

Thank you

tony1ant, 2020-09-07:
#oldman4U 

Sorry I should have put Ender 3 instead of E3, Please accept my Apologies. I'm new to GitHub & still learning. I use voice to text because I have Rheumatoid Arthritis, Fibromyalgia & CRPS, because typing is very difficult most of the time. In my voice to text I use E3 as a shortened version for Ender 3. 

My PC's PSU died on me so I haven't been able to try the suggestions given 2 me. As soon as I receive my new PSU I will try again. Please note in my firmware there is no 'E3' in it was I said Typonese lol. Thank U 4 your reply. 

Thank You & Hava Lovely Day
Kind Regards Tony

oldman4U, 2020-09-07:
Hi Tony.

It is all good - and because of your ticket traffic-light improved his "Read Me" and included the description from above.

We are all working together so we all can enjoy our hobby.

Let us know

;-)

oldman4U, 2020-09-10:
Hi.

Any news?

tony1ant, 2020-09-15:
Hi oldman4U 
My PSU blew up on my PC. So I'm waiting for a new one. Once I get my PC up and running hopefully this week. I will try the suggestions & then let you know. Thank you again 4 your support & hard work.

oldman4U, 2020-09-15:
Thank you for letting us know 

👍🏻

tony1ant, 2020-09-15:
Hi Every1 
I finally got my PC going again. I've tried the suggestions and now I have even Less Hair. I still get nothing on the Marlin screen. Touchscreen works perfectly. I'm new to VSCode and Github so Please excuse my ignorance. I'll attach my configuration files. Thank You.

[configuration.h.txt](https://github.com/bigtreetech/BIGTREETECH-TFT35-V3.0/files/5228468/configuration.h.txt)
[configuration.adv.h.txt](https://github.com/bigtreetech/BIGTREETECH-TFT35-V3.0/files/5228469/configuration.adv.h.txt)
[image]

oldman4U, 2020-09-16:
Hi.

Can you quit VSCode and try it again.

Please attach your config files as such and not as textile.

Thank you

oldman4U, 2020-09-16:
I also tested possible reasons for getting the LCD 12864 emulator only and one possible reason is the cable. I get exactly this once i remove the cable from a perfectly working system.

tony1ant, 2020-09-16:
Hi oldman4U 
I tried to attach the config.h files as config.h but it says this type of file is not supported. I'm sorry 4 being such a pain. I will try again. 

I suffer with CRPS, Arthritis Fibromyalgia & other problems, which give me constant excruciating levels of pain. I got into 3D printing 2 distract my mind from the 24/7-365 constant pain. So Please bear with me if I don't answer straight away.

tony1ant, 2020-09-16:
[image]

this is the error I keep getting . I just noticed that .zip files R accepted so I will try that now. Thanks.

tony1ant, 2020-09-16:
Hi Every1
[Files 4 Upload 2 Github.zip](https://github.com/bigtreetech/BIGTREETECH-TFT35-V3.0/files/5233615/Files.4.Upload.2.Github.zip)
I think I've done it right so here is the .zip file with my config.h, config.adv.h, platformio.ini and firmware.bin

BobP2020, 2020-09-16:
I looked at your configuration.h file. As I mentioned earlier, you should comment out (add // in front of the line of code) reference to the "#define CR10_STOCKDISPLAY" as you are not using this display as you said you have a TFT35 E3 display connected.
Next, you must "un-comment" this line for the  "#define REPRAP_DISCOUNT_FULL_GRAPHIC_SMART_CONTROLLER" if you want the 12864 emulation to work. So, add the // at beginning of line 2083 and remove the // at the beginning of line 1997. These line numbers come from the file you recently uploaded.

oldman4U, 2020-09-16:
Tony. There is nothing to excuse. 

@BobP2020 could you pls chsnge it and attach it here. I am on my mobile. Thank you

BobP2020, 2020-09-16:
> 
> 
> Tony. There is nothing to excuse.
> 
> @BobP2020 could you pls chsnge it and attach it here. I am on my mobile. Thank you

I am not sure what it is that you would like me to change and attach...Tony's Configuration.h file ?

BobP2020, 2020-09-16:
[16Sep20_Configuration.zip](https://github.com/bigtreetech/BIGTREETECH-TFT35-V3.0/files/5234075/16Sep20_Configuration.zip)
OK I zipped the modified Configuration.h file that to avoid confusion on my part, I prefixed with a date-code of 16Sep20_. Just unzip and edit the filename to remove my date code. 
I hope that is what was being asked for and I hope it offers some sort of help.

BobP2020, 2020-09-16:
As an additional comment before I have to leave my computer, double check that the two gray cables connecting the TFT35E3 display to the SKR 1.4Turbo board are in the EXP-1 and EXP-2 sockets. On the display these are the two sockets closest to the SD socket. Of course, EXP-1 will go to EXP-1 on the SKR1.4 etc.  Just to double check as some folks have connected to the EXP-3 connector but that is just for when using the stock display, NOT for when using the TFT35.

oldman4U, 2020-09-16:
@BobP2020 Thank you for your help, i really appreciate it!

@tony1ant Maybe you can attach a picture of the cables and where they are connected.
Today i got an SKR 1.4 which i bought so i can help better. Attached are the cables how they have to be connected. With BobP2020's configuration files you should be able to compile the firmware right.

Let us know

Regards

oldman4U, 2020-09-16:
Evergreen;-)

Forgot to attach the picture...

[image]

tony1ant, 2020-09-16:
Here R the fotos
[image]
[image]
[image]
Thank U.

:

BobP2020, 2020-09-16:
OK, confirmed that you do not need to use the EXP-3 for the setup that you have. I did notice in your photos that you are not using the cables that came with the display (EXP-1 and EXP-2). Have you confirmed that your replacement cables are giving proper connections? Can you temporarily substitute back the cables that came with the display? It would be a good trouble-shooting test. This is all in an effort to maintain a certain amount of hair on your head LOL.

BobP2020, 2020-09-16:
> 
> 
> I tried disconnecting EXP-3, then it said 'No Printer Connected' and still no Marlin screen. Now I get the error below:
> #include errors detected. Please update your includePath. Squiggles are disabled for this translation unit (K:\Marlin\src\MarlinCore.cpp).

The "No Printer Connected" appears to be a message the touch-screen mode puts out if it cannot make a serial connection to the control board. Just to be sure we are talking about the same connections, EXP-1, EXP-2 are the cables needed for "Marlin" mode A.K.A. 12864 Emulation. The "RS232" cable is needed for proper touch-screen operation. So in a final setup where both modes are supported there will be a total of 3 cables running from the display to the control board. 

What was changed that caused the error message listed above? There are some "error type" messages that can be ignored as long as your compile is a "SUCCESS". I know I have had to ignore the "Squiggles" message and I still get a good compile. There is a LOT going on within PlatforIO that only savvy programmers understand

tony1ant, 2020-09-16:
The "No Printer Connected" appears to be a message the touch-screen mode puts out if it cannot make a serial connection to the control board. Just to be sure we are talking about the same connections, EXP-1, EXP-2 are the cables needed for "Marlin" mode A.K.A. 12864 Emulation. The "RS232" cable is needed for proper touch-screen operation. So in a final setup where both modes are supported there will be a total of 3 cables running from the display to the control board.

What was changed that caused the error message listed above? There are some "error type" messages that can be ignored as long as your compile is a "SUCCESS". I know I have had to ignore the "Squiggles" message and I still get a good compile. There is a LOT going on within PlatforIO that only savvy programmers understand.

Thanks 4 the reply. I dont like things beating me & I will try my best 2 persist until I find a fix' with all your help. I've uninstalled VSCode on my PC and reinstalled it. Perhaps some files got corrupted when I added a bigger SSD 2 my PC along with a new Graphics card and PSU this week. I'm gonna try recompile again and I will keep U informed. I've never used VSCode or Github B4, until I got my BTT 1.4Turbo board etc 4 my Ender 3 Pro. I am definitely Not a savvy programmer, but I hav been building my own PC's 4 many decades. I'm very grateful 4 all your time 2 help me. Hopefully I can return the favour 1day. Thank U.

oldman4U, 2020-09-17:
Please let us know. 

Thank you

tony1ant, 2020-09-17:
#BobP2020
Hi Bob, I connected the original cables that came with the screen & still No Luck. I checked them for continuity with my new Multimeter-Oscilloscope and they are Ok. The Touchscreen works perfectly & both nozzle & bed heat up as they should. I tried to recompile so I can add my NeoPixels, BL-TOUCH & Filament Runout Sensor, but they failed. I decided to start from scratch again, downloaded all the files I needed again, when compiling the platformIO.ini and it failed in VSCode. Notepad++ was so much easier & so was ArduinoIDE. I'm so tempted now to put back my original Creality E3 Pro board which printed perfectly, but it is so noisy compared to the new BTT SKR1.4 Turbo board & TMC2209 stepper drivers. I've tried resetting the board back to default, still no joy, No Emulated Marlin screen only the text 'LCD12864 Simulator'. I'm really puzzled now :((

BobP2020, 2020-09-17:
Hmm, this is a real mystery as I am using similar hardware (as far as the display and control board) and after a lot of  hours/days in the learning process (making mistakes, that is haha) I have been running in dual mode for almost a year now. The only things that I can think of at the moment have to do with your definition of serial ports in Configuration.h and the #define CONFIG_EXAMPLES_DIR I mentioned earlier (it appeared on line 24 of the config.h file you submitted originally). 

The only thing I can suggest is to comment out the code line dealing with the " #define CONFIG_EXAMPLES_DIR". This code line will appear near the beginning of the Configuration.h file ...AND... the Configuration_adv.h file.  To be honest, I could not find out what that line of code was supposed to do.  My configuration files have no such entry but then my printer is a DIY build.
Also, comment out the code line "#define SERIAL_PORT_3 1". 
The only reason I make these suggestions is because they do not exist in my Configuration.h file. There is the possibility that commenting out the serial port 3 might cause something else to not work (such as WiFi if you were trying to use that) but this would put our configurations on a common ground...just to get the display to work properly. BTW-I feel the effort to get the SKR1.4T with the 2209 is going to be worth it in the end and we both will be a lot smarter LOL.
Another suggestion I would like to make is to start simple..that is, do not try to bring on any other feature until the display is working correctly. I have read that there can be some unusual interactions between various features associated with the BTT systems but they DO work. Hoping for your success so keep us posted.

tony1ant, 2020-09-17:
Hiya Bob

Great minds think alike lol. U took the words right out my mouth. I just about to disconnect everything and only connect the screen. The rest can wait until we've solved this puzzle. And Yes we will certainly B much wiser. I've emailed BTT 2day also 2 C what they come up with. Maybe the screen is faulty? I've been itching 2 get it going since it arrived. I suppose the old saying 'All Good Things Come To Those Who Wait' & being Wheelchair bound-Disabled, I hava lot of patience & tolerance.
Thanks again Every1 for your help. I will keep U posted.

oldman4U, 2020-09-19:
Hi. Short update. 

I tester the additional serial port 3 which you defined to see it it makes a difference, but both "screens" still worked. Could you please attach your two configuration files again. Thank you

tony1ant, 2020-09-21:
Sorry that I haven't replied sooner I had a very bad pain day yesterday. Here are the files. Thank U.
[Config.Files4Github.zip](https://github.com/bigtreetech/BIGTREETECH-TFT35-V3.0/files/5253625/Config.Files4Github.zip)

oldman4U, 2020-09-21:
Dear Tony.

There are many other users which do not answer for weeks and months and once you ask them to close the ticket they complain about it. So don't worry if you can not answer for a few days, it is all good.

oldman4U, 2020-09-21:
Good news my friend.

Using your SKR firmware i have the same issue, so there is a problem with the firmware.

oldman4U, 2020-09-21:
Hi.

I am not allowed to send you a compiled version directly here using this GitHub. So do you have a eMail address or an upload link for me, so I can send you the firmware. I believe it will work once you updated the firmware of your SKR.

Let me know. thank you

oldman4U, 2020-09-21:
Can you send me a eMail address or a Upload link, so I can send you the firmware. It is not allowed to send you the firmware directly - unfortunately.

tony1ant, 2020-09-21:
Hiya oldman4U
Where should I send U my email, ? 
I don't want it public because of spam. I don't C anything here where I can send U a msg directly.
Thank U.👍👍👍

tony1ant, 2020-09-21:
Hiya oldman4U

Here is the link 4 my email. 
https://www.dropbox.com/s/b0r0224a7cs6aui/Hiya%20oldman4U-My%20Email.doc?dl=0

Could U Please let me know if U've got it?. Cheers Mate.

Thank You & Hava Lovely Day
Kind Regards Tony

tony1ant, 2020-09-21:
Hi Every1
I've also emailed my firmware files 2 BTT so they can also hava look & C Y the Marlin Screen isn't working. But I dunno when I'll get a reply from them cause they said they hava long backlog of emails 2 reply 2.
Thanks again 4 all your help, I'm Eternally Grateful.
👍👍👍

oldman4U, 2020-09-22:
Maybe it would be best to make a gmail address which you use only for such
things.

oldman4U, 2020-09-22:
Hi.

You can delete the link from above.

Firmware is on the way.

;-)

tony1ant, 2020-09-22:
Thank you very much. How do I delete the link?

oldman4U, 2020-09-22:
It is possible to edit a text.

tony1ant, 2020-09-22:
Very strange on the GitHub app on Android there no options to edit or delete, but when I opened it up in Chrome the options are there. I do hava Gmail email address. If it's easier I'll use that 1 in future. I will install the firmware you sent later on & I will let you know the outcome. Thank you oldman4U 👍👍👍

tony1ant, 2020-09-22:
Hiya Every1
Guess What...........?

oldman4U, 2020-09-22:
It works....😎

oldman4U, 2020-09-22:
The GitHub app on iOS is a nightmare and crashes frequently.

tony1ant, 2020-09-22:
I don't like CrApple products, I used 2 repair them every day, when I was able 2 work. I'm a Android fan 😁😁😁

**Nobody guessed...?....** 

_Well...... Hava Look @ these pics_:

[image][image][image][image]

☆🌟☆🌟☆🌟☆🌟☆🌟☆🌟☆
╔══╦╗──────╔╗
╚╗╔╣╚╦═╗╔═╦╣╠╗
─║║║║║╬╚╣║║║═╣
─╚╝╚╩╩══╩╩═╩╩╝
╔═╦╗  °☆°☆°☆°☆°☆°
╚╗║╠═╦╦╗        ☆°
╔╩╗║╬║║║      ☆°
╚══╩═╩═╝    ☆°
🌟☆🌟☆🌟☆°☆🌟 So Much Every1 4all your help, now maybe my hair will grow back.😂😂😂

#oldman4U U R a Star, I installed the files and hey Presto I now have both Marlin & Touchscreen.
I'm Eternally Grateful.

Thank You & Hava Lovely Day
Kind Regards Tony

oldman4U, 2020-09-22:
Last week I updated my iPhone 6s which i bought almost 5 years ago to the latest iOS v14. To transfer the data from an old to a new iPhone you have to turn the new one on, select the language, select the country and scan a QR code. We have 10 iPhones in my family and the amount of time I need to spend to maintain them is almost zero.

Finally I am happy to hear that the firmware update solved the problem. Even I changed the firmware line by line, it is possible that i missed one or two settings, so please check them. BLTouch was not activated and is still not activated.

One of the problems with the firmware you sent to me, was the fact, that the display was still wrong. CR-10 instead of RepRap was defined, like @BobP2020 wrote 16 days ago.... But finally I always got compile errors using it, so i transferred everything to the version form yesterday and compiled it new. So several problem, some of them unknown.

I hope you have fun and ask you to close the ticket once you do not need it anymore,

Thank you and happy printing

BobP2020, 2020-09-22:
I think the reason the display issue regarding the definition of the stock CR-10 display showing up comes from the use of the example configuration. I personally prefer to edit the configuration files directly an not use the example files except for a reference...but that is just me. I think the reference in the configuration files to the Ender3-Pro directory kept feeding in the wrong display. Because of the modifications being made I think it would be better to not use that reference, I think it may have been the root cause of the problem but someone more familiar with exactly how the example configuration file is supposed to work would have to comment on this aspect. All is well that ends well LOL.

oldman4U, 2020-09-22:
Good point. I never used the configuration files and always use the latest bugfix and sometimes for new features the config for reference - like you.

Finally Marlin can not know which additional hardware is attached and the TFT firmware can not know what's and how attached on the connectors and if this is right. And making a sample config for every single case is impossible. Imagine how many factors are available resulting in how many possible configuration...???

Describing the functions is a first step. For the TFT/mainboard/SD card issue it would maybe be best to make a kind of function tree which starts with:

A: You have a TFT with NO Encoder (TFT28 for example) - next point A1:
B: You have a TFT with Encoder (TFT 24/35.... and so on next Point B1:

A1: You would like to use the Marlin emulator mode - YOU ARE SCREWED;-)))) HAHAHA
B1: ......

;-)

tony1ant, 2020-09-23:
Hiya Every1
Well I have certainly learned a lot from all of you & in future I'll do the same and only use the examples as a reference. I've only been doing firmware upgrades for my CE3 Pro in the last 8 months or so mostly in ArduinoIDE & Notepad++. VSCode is a whole new kettle of fish LoL, but I'm learning albeit slowly 😁😁😁. Thank you again 4 all your help. I'm sure I'll be finding other issues as I upgrade my 3D Ender 3 Pro & my son's Creality CP-01 (he only 11 & in high school already). I'm so looking forward to finishing up tidying up the cables etc & doing my 1st print with the new SKR 1.4Turbo mainboard &  TMC2209 stepper drivers etc. I'll post pics on Thingiverse, Fbook & the others after its all done. This issue is n now resolved, should I now close this issue?
Thank You Very Much, I have an Ocean of Gratitude. 👍⭐⭐⭐⭐⭐👍

oldman4U, 2020-09-23:
It was my pleasure to help you. 

Thank you @BobP2020 

Yes. Please close this ticket as it is solved. Start a new one under the corresponding repository whenever needed. 

Cheers

tony1ant, 2020-09-23:
Thanks again Every1. I think the main problem is I don't know how to compile Marlin 2.0 correctly in VSCode (I was editing the config.h files, as  BobP2020 suggested in the beginning, then saving them from the file menu and then i copied the supposedly edited files (which I thought had been changed as oldman4U mentioned that my BL-Touch isn't enabled even though I thought it was enabled). Whenever I clicked the ✔️ to compile in VSCode it failed, so I saved it from the file menu save option which obviously didn't compile the files. I assume the two are not the same thing?   I need to learn how to compile & use VSCode properly. 
👍👍👍Thanks again guys & Stay Safe my good friends👍👍👍

closed 2020-09-23 by tony1ant

oldman4U, 2020-09-23:
Hi. 

I could not compile your code either but the one I sent to you should work. Check it out😁

Edwin0011, 2020-11-23:
What was the Fix????  I have same Issue TFT35 e3 V3.0 
My touch screen works fine I hold for 3 seconds and switch to LCD12864 Simulator..
LCD12864 Simulator just shows on screen.
I turn knob nothing.
I have #define REPRAP_DISCOUNT_FULL_GRAPHIC_SMART_CONTROLLER  uncommitted
I have no other controllers uncommitted

thanks in advance

oldman4U, 2020-11-23:
The fix is to connect the cables right and to choose the right display in Marlin.

Which mainboard?
Send pictures showing the cables.

Edwin0011, 2020-11-23:
so ya uhm Exp1 & Exp 2 were backwards LOL