bigtreetech/BIGTREETECH-TFT35-V3.0 #83: No RSET for TFT on MKS SGEN L v2.0

ZhuDaHai, 2020-09-10:
AUX-1 on the MKS SGEN L v2.0 mainboard has 5v G TX and RX, but no RSET (reset) for hooking up the TFT connector cable from the BTT TFT35 v3.0.
Is RSET required, and if so, where to plug it on that board?

Thanks!

oldman4U, 2020-09-10:
RST is not required and if the mainboard does not support it, you will not be able to connect it. Isn't this more a question for MKS?

ZhuDaHai, 2020-09-11:
> RST is not required and if the mainboard does not support it, you will not be able to connect it. Isn't this more a question for MKS?

MKS didn't produce a non-standard serial connector with poor to non-existent documentation. But, hey, they're supposed to know how to interface to every display board out their while BTT can emulate Apple...

Nevertheless, I got my answer direct from MKS Support within a few hours as their customer service actually is Customer Service.

closed 2020-09-11 by ZhuDaHai

oldman4U, 2020-09-11:
Thats the difference between open source and a solution where everything
comes from a single manufacturer.

👍🏻