bigtreetech/BIGTREETECH-TFT35-V3.0 #84: M109 to M104 Conversion Prevents Neopixel Status LEDS

avidNewb, 2020-09-24:
When an M109 is sent to the printer through the TFT it is converted to an M104. This prevents the Status LED feature in Marlin from being triggered. Would it be possible to implement a similar feature through M150 in the TFT?

oldman4U, 2020-09-24:
Hi.

I would recommend to ask this question in the generic firmware section which is covered by more users, even more, because this seems to be not really hardware specific.

Best would be also to define better what you mean with "send".

Please close this ticket once you do not need it anymore.

Thank you

oldman4U, 2020-09-27:
Please help the community and close the ticket once you do not need it anymore. Thank you

avidNewb, 2020-09-27:
I will recreate as requested when I have a chance.

closed 2020-09-27 by avidNewb