bigtreetech/BIGTREETECH-TFT35-V3.0 #86: Touchscreen doesn’t respond and screen is not touching the frame where it’s mounted in.

jlogin2002, 2020-10-11:
Touchscreen doesn’t respond and screen is not touching the frame where it’s mounted in.

After shiwch on first time, it enter into calibration mode but it dos not work. I powered off and powered on again and calibration mode does nos show any more. 

Touch screen does not work. Emulated terminal is working

I trayed to upload a new firm but it remais the same,

I look forwaard to hearing from your news soon

Regards

oldman4U, 2020-10-11:
So you have the same issue with two different TFTs?

Close the ticket which does not match your TFT.

jlogin2002, 2020-10-11:
This is what you asked me:

This is a brand new TFT or has it been working before?:  **YEs it is a brand new**
Which mainboard? BTT GTR V1.1. MArlin has being configurated an loaded and it works. #define REPRAP_DISCOUNT_FULL_GRAPHIC_SMART_CONTROLLER has been activated int .h file
What exactly is connected on which connector of the TFT? **Black cable into TFT conector at the main board at the right position**
Which firmware and have you updated the firmware already?  **BIGTREE_TFT35_V3.0.26.x.bin+ config+TFT35 folder**
Does it work with only the RS232 cable connected? ** LCD12864 simulator works, BTT touch panel is dispalyed with icons  but does not work**

Thank you in advance

oldman4U, 2020-10-11:
Add an empty textfile named Reset.txt to the SD card and restart the TFT.

jlogin2002, 2020-10-11:
Hi Oldman

It works....Great

Thank you very much for your quick answer!

closed 2020-10-11 by jlogin2002

oldman4U, 2020-10-11:
👍🏻

Happy printing