bigtreetech/BIGTREETECH-TFT35-V3.0 #87: My led knob dont changes color

yousssert, 2020-10-17:
Cant someone help me

yousssert, 2020-10-17:
https://www.reddit.com/user/YousefAlHassan/comments/jd0jid/my_tft35_e3_v3_led_knob_dont_work/?utm_medium=android_app&utm_source=share

oldman4U, 2020-10-17:
Hi. Which firmware is installed and shown under Information?

yousssert, 2020-10-18:
> Hi. Which firmware is installed and shown under Information?

TFT35_E3_V3. 0.26.X Sep 12 2020

yousssert, 2020-10-18:
> Hi. Which firmware is installed and shown under Information?
And marlin is 2.0.7.2

oldman4U, 2020-10-18:
Please install the latest Master firmware for the TFT according to the ReadMe document which you can find here: https://github.com/bigtreetech/BIGTREETECH-TouchScreenFirmware

Please close this ticket once the problem is solved.

Thank you

oldman4U, 2020-10-26:
Hello?

It is very unfriendly not to answer after receiving help.

thebiggestgit, 2020-12-23:
Stick on the latest firmware, make sure you use the E3 version. I'd not used the E3 version accidentally when I updated and although the LED colour options were on the graphical menus they weren't under the settings and didn't work as wrong firmware.