bigtreetech/BIGTREETECH-TFT35-V3.0 #88: TFT35 v3.0 and SKR Mini E3 - Can't select mode

DiasWindu, 2020-10-23:
I have just upgraded my ender 3 pro with the TFT 35 v3.0 and the SKT Mini E3 v2.0

Currently the TFT35 is NOT mounted. It is connected to EXP3 on the TFT. The RS232 is connected correctly. I have updated to the latest firmware on both the SKR Mini and TFT. When updating the firmware, the touchscreen calibration initiated and functioned correctly. 

On start up I get to the screen to select Marlin or Touch mode, but it will not allow me to select a mode. I have tried using the control knob, and also using the touchscreen. I have read you need to press and hold the knob for 10 seconds to enable touch screen, but this does not work.

I have reset the TFT by placing an empty file called Reset.txt on my SD card. This brought up the touchscreen calibration again which worked fine, but I am still stuck unable to select a mode.

Once again the TFT is currently NOT mounted so this is not the issue of the screen pressing against the metal case. 

Can anyone offer a solution please?

oldman4U, 2020-10-23:
Hi. So you defined touchmode as default in config.ini.

Then you press the knob for 3 sec and the selection appears. But from here you can not enter Marlin mode because nothing happens when you press Marlin? In touch mode you are able to navigate in the menus?

DiasWindu, 2020-10-23:
Hi, Thanks for the reply.

Yes I have set touchscreen as default. 

When I power on the printer I am getting the screen to select Marlin or Touch. 

It will not allow me to select an option with the control knob or the touchscreen.

oldman4U, 2020-10-23:
But you can navigate in menus of the touchscreen?

DiasWindu, 2020-10-23:
No. 

I only see the screen to select touch mode or marlin mode. I am unable to choose one, and cannot move past this screen.

oldman4U, 2020-10-23:
Sorry, got it....now;-(

Have you tried to connect only one cable (EXP3 or RS232)?
Anything else connected to the TFT?

DiasWindu, 2020-10-25:
Thankyou for trying to help.

There is nothing else connected to the TFT. I have tried using only the EXP 3, and also only the rs232 connected with the same problem

DiasWindu, 2020-10-25:
There has been no luck on my end. I think I may have received a faulty board

oldman4U, 2020-10-25:
I would update to latest master branch firmware and try to change default mode in config.ini.

In addition you can try to cut an old usb cable and connect the red and black strip to 5V and ground of the rs232 connector.

You can also try to remove the reset strip from the standard cable.

If this all does not help contact your dealer for exchange.

DiasWindu, 2020-10-27:
Thanks for the advice. 

Sadly it did not work so I have emailed my dealer.

oldman4U, 2020-10-27:
Makes sense 😔

Please help the community and close the ticket once you do not need it anymore. Thank you

closed 2020-10-27 by DiasWindu

gallowayk, 2022-02-11:
Not sure why this issue was closed out if it was unresolved. I'm not experiencing the exact same issue and cannot figure out, for the life of me, why it doesn't work properly.

oldman4U, 2022-02-11:
hi.

why should it be left opened?

If you HAVE the same issue, you may start a ticket in one of the BTT user groups. This is the BTT TFT firmware repository, so if you believe a firmware related issue causes this you are welcome to start a new ticket here.