bigtreetech/BIGTREETECH-TFT35-V3.0 #90: tft35 v3 errors

Kscott73, 2020-11-07:
firmware: tft35_v3.0.26 apr 11 2020
button pics  are labeled wrong  and coordinates are wrong - home: is 300,310,400
(moves in the direction of button pics

[image]

oldman4U, 2020-11-07:
Hi.

Do you think this is a hardware issue?

Kscott73, 2020-11-08:
dont think so, movement is correct to button pics, button labels are wrong

oldman4U, 2020-11-08:
Software related issues should be started using the software related repository. Update to a firmware from this week and see if this solves the problem.

oldman4U, 2020-11-08:
This is how it looks with an actual firmware.

[image]

Kscott73, 2020-11-09:
tried the new firmware, didn't work
the x and z labels are incorrect, along with the coordinates
when press home, it moves in right direction but coordinates are 300,300,400

oldman4U, 2020-11-09:
Attach the config.ini (.CUR), which you have been uploading to the TFT.

oldman4U, 2020-11-09:
The problem is, that you have selected Classic Mode in Config.ini and not Unified Mode like I have. Changing this is config.ini will solve the problem.

Kscott73, 2020-11-09:
firmware uploaded, it did save.. i changed the logo bmp
problems are still there

[config.zip](https://github.com/bigtreetech/BIGTREETECH-TFT35-V3.0/files/5511355/config.zip)

oldman4U, 2020-11-09:
Take a picture from the main screen, from the Move screen and the Info screen and attach it here, please.

oldman4U, 2020-11-09:
And you sent me 2 files with the extension .CUR, but the config is still .ini... Has the config.ini been updated during firmware update?

Kscott73, 2020-11-09:
[image]
[image]
[image]

oldman4U, 2020-11-09:
The firmware is older than old.

Read the ReadMe at this page: https://github.com/bigtreetech/BIGTREETECH-TouchScreenFirmware

Do what's described there and let me know.

Kscott73, 2020-11-09:
i've deleted all tft files, zip files
re-downloaded https://github.com/bigtreetech/BIGTREETECH-TouchScreenFirmware.git
i've changed to classic bmp&fonts
sd card:
[BIGTREE_TFT35_V3.0.26.x.zip](https://github.com/bigtreetech/BIGTREETECH-TFT35-V3.0/files/5513046/BIGTREE_TFT35_V3.0.26.x.zip)

Kscott73, 2020-11-09:
it started off great.
error:
found outdated data:
insert the SD with the required files and press reset to update

oldman4U, 2020-11-09:
Outdated data is a good sign.

Make a picture and attach it or check with data is outdated on screen.

oldman4U, 2020-11-09:
The content of your SD card looks good in general.

Kscott73, 2020-11-09:
[image]

Kscott73, 2020-11-09:
why am i getting outdated files, im downloading from straight from github
[BIGTREE_TFT35_V3.0.26.x.zip](https://github.com/bigtreetech/BIGTREETECH-TFT35-V3.0/files/5513374/BIGTREE_TFT35_V3.0.26.x.zip)
here is the cur files

oldman4U, 2020-11-09:
Uii, this looks strange.

It should look a bit like this:
[image]

Your .CUR files look good, so everything which can be updated has been updated, but somehow this ended in a corrupted firmware.

I believe the precompiled firmware is somehow corrupted. Please take the attached file, unzip it and change the TXT to bin. Put it on the SD Card and update the firmware again.

Please let me know

[BIGTREE_TFT35_V3.0.26.x.TXT.zip](https://github.com/bigtreetech/BIGTREETECH-TFT35-V3.0/files/5513501/BIGTREE_TFT35_V3.0.26.x.TXT.zip)

Kscott73, 2020-11-10:
OK
1) Format SD
2) Unzip
3) Copy File & Rename Extension
4) Eject SD

I now have a bigger problem
[TFT35.zip](https://github.com/bigtreetech/BIGTREETECH-TFT35-V3.0/files/5514002/TFT35.zip)

(https://github.com/bigtreetech/BIGTREETECH-TFT35-V3.0/files/5513986/TFT35.zip)
[image]

Kscott73, 2020-11-10:
last try before calling it a night
used your file
[BIGTREE_TFT35_V3.0.26.x.zip](https://github.com/bigtreetech/BIGTREETECH-TFT35-V3.0/files/5514244/BIGTREE_TFT35_V3.0.26.x.zip)
can't use touch screen and switching other mode is just a black screen
[image]

oldman4U, 2020-11-10:
Make a textfile named reset.txt, put it on sd card insert into tft and press reset knob.

Kscott73, 2020-11-10:
yes, i have touch screen and more buttons..
now its just stallguard giving me problems
thank you

[image]

oldman4U, 2020-11-11:
The firmware is still old, maybe you want to update it.

Please close this ticket, thank you.

closed 2020-11-13 by Kscott73