bigtreetech/BIGTREETECH-TFT35-V3.0 #91: no connection to the btt skr 1.4 turbo card

dacentichka, 2020-11-09:
good day, I'm from Russia, I write via Google translator. I have a tft 35v3 and skr 1.4 turbo mode 12864 everything is good but in sensor mode the screen can not control the printer and says there is no connection, I tried all firmware screens ranging from 0.24 to 0.26, all Assembly Marlin for this Board, I tried to swap the ports and put a different speed in the firmware but it gave no result, but when I connect Reiterhof that appears on the screen temperature sensors and writes the printer ready. please tell me what to do?

oldman4U, 2020-11-09:
Hi.

Please take pictures showing the cable and attach your Marlin and TFT configuration files here.

Thank you

dacentichka, 2020-11-09:
> Привет.
> 
> Пожалуйста, сфотографируйте кабель и прикрепите сюда свои конфигурационные файлы Marlin и TFT.
> 
> Спасибо!

[image]
[image]
[image]
[image]
[image]
[image]

dacentichka, 2020-11-09:
[Uploading firmware_tft 35.zip…]()

dacentichka, 2020-11-09:
[Uploading firmware_skr_1_4_turbo.zip…]()

dacentichka, 2020-11-09:
> Привет.
> 
> Пожалуйста, сфотографируйте кабель и прикрепите сюда свои конфигурационные файлы Marlin и TFT.
> 
> Спасибо!

oldman4U, 2020-11-09:
Problem solved?

dacentichka, 2020-11-09:
> Проблема решена?

no, I didn't click this close button on my phone, I don't know English very well, I didn't know that this would close the question, and I don't know how to open it again

oldman4U, 2020-11-09:
No problem! Your english is good enough.

Have you tested the cable before you extended it or do you have a second cable to test?

Please also make a picture which shows the problem "Printer not attached".

oldman4U, 2020-11-09:
And when you upload something, please wait to press the "Comment" button until the Upload is finished.

Please upload the firmware again.

Thank you

dacentichka, 2020-11-09:
> И когда вы загружаете что-то, пожалуйста, подождите, чтобы нажать кнопку "комментарий", пока загрузка не будет завершена.
> 
> Пожалуйста, загрузите прошивку еще раз.
> 
> Спасибо!

I'm still just learning to use github, I tested the wire, and I also tested the contacts on the boards with the wire in order to eliminate the pin breakage in the plug, everything is fine there, and I downloaded the firmware from your github, opened the skr 1.4 turbo folder in the visual Studio, checked the serial ports and clicked compile without changing anything else, since I'm just wondering how it works before I put it in a working printer, I can't put it on the printer until everything works as I need to print a lot

dacentichka, 2020-11-09:
[image]

oldman4U, 2020-11-09:
[image]

I can not see the Uploaded files, only the first Pictures;-(

dacentichka, 2020-11-10:
> [image]
> 
> I can not see the Uploaded files, only the first Pictures;-(

dacentichka, 2020-11-10:
[firmware_skr_1_4_turbo.zip](https://github.com/bigtreetech/BIGTREETECH-TFT35-V3.0/files/5514435/firmware_skr_1_4_turbo.zip)

dacentichka, 2020-11-10:
[image]

dacentichka, 2020-11-10:
I could not attach the entire folder of Marlin because it can not attach more than 10mb, I can send the entire archive to the mail

oldman4U, 2020-11-10:
So the update was successful. Great. 

Can you navigate to Info and take a picture.

dacentichka, 2020-11-10:
are you talking about the information in the minu on the display or in the information in the serial port on the M501 command?

oldman4U, 2020-11-10:
First.

dacentichka, 2020-11-10:
[image]
[image]

dacentichka, 2020-11-10:
> First.

so what's next?

dacentichka, 2020-11-10:
> Обновление до самой последней прошивки. Это с апреля, и есть один доступный со вчерашнего дня.

I'll try it now

dacentichka, 2020-11-10:
> Update to the most recent firmware. This is from April and there is one available from yesterday.
I couldn't flash from the first folder, but from "Copy to SD Card root directory to update - Unified Menu Material theme" everything worked.

dacentichka, 2020-11-10:
[image]
[image]

dacentichka, 2020-11-10:
[image]

oldman4U, 2020-11-10:
Make sure you have the following settings defined in Marlin:

#define SERIAL_PORT 0
#define BAUDRATE 115200

and also 115200 defined as Baudrate for the TFT under Menu-Settings-Connection-Baudrate

dacentichka, 2020-11-11:
please tell me, and this should be viewed in the config.h or in config. adv. h? and another question, I tried to build the firmware for the screen myself, but it immediately gives an error when compiling that I don't understand, I took a screenshot of the error, tell me what to do please.
[image]

oldman4U, 2020-11-12:
The settings i wrote above are all in Configuration.h of your mainboard.

Can you compile other firmware. Looks like a problem with VSC/PIO. Rester VSC and see if this solves the problem.

dacentichka, 2020-11-12:
should port "0" be port number one or port number two?

 I'll try to reinstall the VSC.

oldman4U, 2020-11-12:
#define SERIAL_PORT 0

Like I wrote it, it can only be the first entry otherwise i would have written:

#define SERIAL_PORT_2 x

dacentichka, 2020-11-12:
so I also tried, ie port 2 is commented out, and the serial port place -1 set 0, and also does not work

oldman4U, 2020-11-12:
And the baud rate of mainboard and tft?

dacentichka, 2020-11-12:
I tried everything possible, I will try today in 5 hours as I come home from work to reinstall vsc and compile the screen firmware myself.

oldman4U, 2020-11-12:
You can change the baud rate of the screen in the TFT menu itself. No need to compile the firmware for this. After changing the baud rate, you have to wait 5-10 seconds for the new setting to take over.

bttskrfan, 2020-11-12:
I have the same problem. I found that attaching SKR 1.4 Turbo to Creality Ender 5 LCD using EXP1 I get first Ender 5 logo then Marlin logo 2.0.1.6 and it gets stuck. TFT35 just keeps showing No Printer Attached and black screen when switching to emulator. Please advice.

oldman4U, 2020-11-12:
Correct cabling could help.

https://github.com/bigtreetech/BIGTREETECH-TouchScreenFirmware/issues/1047

[README.pdf](https://github.com/bigtreetech/BIGTREETECH-TFT35-V3.0/files/5532401/README.pdf)

bttskrfan, 2020-11-13:
Sorry to interfere, I should have opened new issue, but since its related here is what I've managed to find out.

1. I've ordered SKR 1.4 turbo + TFT35 + tmc5610 from AliExpress on September
2. Default TFT firmware was April version
3. TFT35 was keep showing "No Printer Attached" error message
4. I've downloaded, built and updated SKR firmaware (2.0.6.1), but no luck
5. Then I've connected SKR to Ender 5 LCD to discover that SKR gets stuck at Marlin logo "2.0.6.1"
6. I went to Marlin and downloaded latest (2.0.7.2), Hooah! It works, but only with Ender 5 LCD. TFT35 still "No Printer Attached"
7. I've downloaded TFT35 firmware from May, Hooah! Connection established, but emulator still black screen
<I'm here now>
8. None of the latest precompiled TFT35 binaries works properly.
8a. The built has missing BMP files which exists, but uses wrong registry, so I had to rename all file names to be UPPERCASE, check this please.
8b. Touch functionality doesn't work with latest TFT35 binaries, the only version that works with me is May

oldman4U, 2020-11-13:
Send pictures which show the cabeling. 

Use self compiled firmware and let me know then.

oldman4U, 2020-11-21:
Any news?

@bttskrfan So you have touchscreen but no Marlin Emulation? Send a picture which shows how the cables are connected between TFT and SKR.

oldman4U, 2020-11-22:
@dacentichka 

Please reply or close this ticket.

closed 2020-11-30 by dacentichka