bigtreetech/BIGTREETECH-TFT35-V3.0 #92: Freeze when connecting from TFT35_V3_E3 (BTT Surface) to TF on SKR Mini E3 V2

Bastlwastl84, 2020-11-10:
Marlin reboots when i click on Print and selecting the TF-Card plugged in at SKR Mini E3 V2 at the BTT_TFT35_V3_E3
Marlin Bugfix. Same happened with 115200 Baud

[platformio - TFT35.ini.txt](https://github.com/bigtreetech/BIGTREETECH-TFT35-V3.0/files/5520333/platformio.-.TFT35.ini.txt)
[Configuration - TFT.h.txt](https://github.com/bigtreetech/BIGTREETECH-TFT35-V3.0/files/5520340/Configuration.-.TFT.h.txt)
[platformio - SKR Mini.ini.txt](https://github.com/bigtreetech/BIGTREETECH-TFT35-V3.0/files/5520341/platformio.-.SKR.Mini.ini.txt)
[Configuration_adv - SKR Mini.h.txt](https://github.com/bigtreetech/BIGTREETECH-TFT35-V3.0/files/5520344/Configuration_adv.-.SKR.Mini.h.txt)
[Configuration - SKR Mini.h.txt](https://github.com/bigtreetech/BIGTREETECH-TFT35-V3.0/files/5520345/Configuration.-.SKR.Mini.h.txt)
[SanityCheck - SKR Mini.h.txt](https://github.com/bigtreetech/BIGTREETECH-TFT35-V3.0/files/5520349/SanityCheck.-.SKR.Mini.h.txt)

oldman4U, 2020-11-10:
Hi.

Your problem with the SD Card Reader is solved?

Bastlwastl84, 2020-11-11:
nope. i have the option SD-Card, USB and TF-Card at the scree, when i'm selecting the TF at Mainboard, it reboots with message "Error". printing from this card at marlin mode is working.

Bastlwastl84, 2020-11-11:
Here how it happens: 
https://drive.google.com/file/d/18HteBm0ZY0FyuwPcP6JCBf1jDiRHlx0Z/view?usp=drivesdk

oldman4U, 2020-11-12:
No idea what the problem could be. Maybe hardware, cable, card. Software seems to be OK, but I can not verify because I do not have this mainboard.

Sorry

Bastlwastl84, 2020-11-12:
> No idea what the problem could be. Maybe hardware, cable, card. Software seems to be OK, but I can not verify because I do not have this mainboard.
> 
> Sorry

if cable i couldn't use Marlin-Mode and GUI Mode (BTT Surface) and send commands. Cables are OK. Printing from Marlin-Mode and GUI-Mode Printing are working, only crashes if i'm doing this.

oldman4U, 2020-11-13:
Sounds logical. You will have to change parts and see what selves the problem. Not using the latest firmware is not an option.

Bastlwastl84, 2020-11-21:
> Any news?

nowthis bug is gone with newest firmware for TFT and Marlin Bugfix, but having a new problem. If printing from TFT a cylinder for testing spiralize mode and the ttf is at start-screen its stuttering. if i'm in a submenue, only al 3 seconds. if the same file at marlin-mode no stuttering. tried higher Baud and the fast mode for gcode

closed 2020-11-21 by Bastlwastl84

oldman4U, 2020-11-21:
Close this and make a new ticket in the firmware repository.

Bastlwastl84, 2020-11-21:
> […](#)

done