bigtreetech/BIGTREETECH-TFT35-V3.0 #93: TFT35 V3.0 screen displays "Found outdated data:

jayala1213, 2020-11-11:


oldman4U, 2020-11-11:
Which sd card are you using. Size and format?

Are the fonts updated during update?

What if you only load the TFT folder to the card?

jayala1213, 2020-11-11:
> Which sd card are you using. Size and format?
> I'm using a Sandisk 64 gig sd card 
> Are the fonts updated during update?
> no they're not 
> What if you only load the TFT folder to the card?
>Ive tried this, nothing works

oldman4U, 2020-11-12:
You need a 8GB or smaller card for FW update: https://github.com/bigtreetech/BIGTREETECH-TouchScreenFirmware

Please read and follow the instructions.

jayala1213, 2020-11-16:
> Any news?

Yeah I tried a 16 gig sd card because I couldn't find an 8 gig anywhere; it it didn't work. I followed the instructions you linked but still no luck. I'm not sure what else I can try. Any suggestions?

jayala1213, 2020-11-16:
> Any news?

Also I just noticed that the config.ini file I downloaded only says config, could that be a problem?

oldman4U, 2020-11-16:
Read the ReadMe from this page, which describes the update a bit more in detail:

jayala1213, 2020-11-16:
Here is a picture of the screen when I put in the SD card. 

[image]

Here is a picture of the files on the SD card. 

[image]

jayala1213, 2020-11-16:
The files stayed the same before and after the update.

oldman4U, 2020-11-17:
Hmmm. It is clear, that the firmware does not match at least the Fonts and the Config.ini.

Your picture shows, that your operating system does not show the file extension for the three elements. Change your OS so it shows the extensions and send me another picture.

jayala1213, 2020-11-17:
> Hmmm. It is clear, that the firmware does not match at least the Fonts and the Config.ini.
> 
> 
> 
> Your picture shows, that your operating system does not show the file extension for the three elements. Change your OS so it shows the extensions and send me another picture.

What do you mean by change my OS? My motherboard?

Grayfox2005, 2020-11-17:
I don't know if it would make a difference but that is not an E3 board - BIGTREE_TFT35_V3.0_E3.26.x.bin vs BIGTREE_TFT35_V3.0.26.x.bin. You need BIGTREE_TFT35_V3.0.26.x.bin for that board I would think.

oldman4U, 2020-11-17:
The picture shows all 3 files WITHOUT file extension and this is Windows
operating system related.

jayala1213, 2020-11-17:
> I don't know if it would make a difference but that is not an E3 board - BIGTREE_TFT35_V3.0_E3.26.x.bin vs BIGTREE_TFT35_V3.0.26.x.bin. You need BIGTREE_TFT35_V3.0.26.x.bin for that board I would think.

Ok I'll try that

jayala1213, 2020-11-17:
> The picture shows all 3 files WITHOUT file extension and this is Windows
> operating system related.
> 
> 
> > Hmmm. It is clear, that the firmware does not match at least the Fonts and
> > the Config.ini.
> >
> > Your picture shows, that your operating system does not show the file
> > extension for the three elements. Change your OS so it shows the extensions
> > and send me another picture.
> >
> > What do you mean by change my OS? My motherboard?

Ok I get it. Let me do that

oldman4U, 2020-11-17:
@Grayfox2005 you are absolutely right. This is the wrong firmware....

jayala1213, 2020-11-17:
It worked!! Ugh what a relief!! Thank you guys!

jayala1213, 2020-11-17:
Ok so it started updating the screen but it gave me an error message saying that their were like 20 files that weren't being updated. Now I have a different screen that appears when I turn it on. 

[image]

Do you know what this could be?

Grayfox2005, 2020-11-17:
Never seen that I have E3 so I didn't use that firmware. Looks like the top portion of your text is missing. I also put the language packs on my sd so I don't know if it needed it. Are you using the "Copy to SD Card root directory to update - Unified Menu Material theme" folder" for your files for the sd card?

jayala1213, 2020-11-17:
Yes I am using that folder, is that the right one? If you see the other picture I posted the screen are slightly different.

Grayfox2005, 2020-11-17:
OK that is the right one for the colorful icons. The other gives you mono color icons. But are you still getting that outdated data message?

jayala1213, 2020-11-18:
Yeah it just says "found outdated data". I've tried downloading it again but it gives me the same message.

Grayfox2005, 2020-11-18:
Is it adding the .CUR to the end of the file and folder names?

jayala1213, 2020-11-18:
> Is it adding the .CUR to the end of the file and folder names?

Yes all the files have .CUR now

Grayfox2005, 2020-11-18:
Well I've exhausted every avenue I know. I'm just comparing my experience with yours. Mine failed when I attempted to modify the firmware code because it wasn't accepting the changes I made in the config. I got the same issue you had afterwards so I just put the one that I was supposed to put in and installed it and it took, fixing everything.

jayala1213, 2020-11-18:
> Well I've exhausted every avenue I know. I'm just comparing my experience with yours. Mine failed when I attempted to modify the firmware code because it wasn't accepting the changes I made in the [...]

Yeah I understand. I'm not sure what else to do either. I'm gonna keep looking though. Thanks for all your help!

oldman4U, 2020-11-18:
Always update using the correct TFT related firmware, config.ini and TFT folder from the same version.

oldman4U, 2020-11-18:
And only use Unified and not Classic folder.

jayala1213, 2020-11-18:
> Always update using the correct TFT related firmware, config.ini and TFT folder from the same version. 

I noticed that my config file does not have .ini; is that a problem?

oldman4U, 2020-11-18:
Yes.

Look for the original name.

jayala1213, 2020-11-18:
> Yes.
> 
> Look for the original name.
> 
> 
> > Always update using the correct TFT related firmware, config.ini and TFT
> > folder from the same version.
> >
> > I noticed that my config file does not have .ini; is that a problem?

I've been looking everywhere for the original config.ini file but all I can find is config file with no .ini. Do you know where I can find it?

jayala1213, 2020-11-18:
I noticed that the unified material theme file had a config file but it isn't .ini. The other Copy to SD Card Root does have the .ini file. Which one do I use?

Grayfox2005, 2020-11-18:
Did you change the settings in your file browser to show extensions?

jayala1213, 2020-11-18:
Yes here they are. 

[image]

Grayfox2005, 2020-11-18:
ummm no

oldman4U, 2020-11-18:
This is how it looks on my machine after downloading the firmware:
[image]

You should follow the update instructions from here: https://github.com/oldman4U/BIGTREETECH-TouchScreenFirmware - scroll down
using the firmware from here: https://github.com/radek8/BIGTREETECH-TouchScreenFirmware/tree/updateBIN

oldman4U, 2020-11-18:
Check especially the part where I describe the update process and how it looks on TFT screen.

Please let me know

Grayfox2005, 2020-11-18:
Wow! Super instructions there, oldman4U. Way more helpful.

jayala1213, 2020-11-18:
> ummm no... Yeah the file extentions are enabled rather but you need a fresh copy of the install so unzip another copy. That's the ones on your SD card. sorry I keep editing this... this is a poor way to communicate but it gets the job done. Those have been installed already so if those are the correct files to install then remove the .CUR on the end of the files so you can use them again.

Ooohhhh I understand what you mean. Here they are. 

[image]

Grayfox2005, 2020-11-18:
So delete the ones on the SD card and just copy over the files again making sure they are correct and try to reinstall them.

oldman4U, 2020-11-18:
> Wow! Super instructions there, oldman4U. Way more helpful.

Thank you. I was also lost when I did my first firmware update. Hope BTT will merge the instructions and the new firmware from radek8 soon, so more users can do the update by themselves.

Thank you for your help, btw. Are you new to 3D printing?

Grayfox2005, 2020-11-18:
No I print occasionally mainly because I don't print dust collectors. I just print things that I need to fix things or make them work better. I've had 3 printers so far. The one I have now is the Ender 3 and its the first one I've modified electronically. I need them to be strong and accurate and that's done better with better electronics as well as the mechanics. I've had them for 7 years now I'm thinking.

oldman4U, 2020-11-19:
So you are a veteran and I am the newbie.;-)

My son prints a lot ad I do the maintenance of our printers. For me it is more fun to help people and learn new things.

oldman4U, 2020-11-19:
@bigtreetech

Another user fighting with outdated firmware.

It is a shame.

jayala1213, 2020-11-19:
> @bigtreetech
> 
> Another user fighting with outdated firmware.
> 
> It is a shame. 

Really is🤦🏽‍♂️

jayala1213, 2020-11-19:
[image]

It works!! Thank you guys for all your help!

oldman4U, 2020-11-20:
Looks good;-)

Please close this ticket.

Thank you

closed 2020-11-20 by jayala1213

oldman4U, 2020-11-20:
👍🏻

Happy printing

XeroDay99, 2021-03-18:
Found outdated data:

Icons: Update required

Help! lol

K40Ronald, 2023-01-12:
Habe das gleiche Problem nach update Tft35-E3 v3.0
[image]