bigtreetech/BIGTREETECH-TFT35-V3.0 #94: "No printer attached!" Error with btt SKR 1.4 Turbo

BlackMalakDown, 2020-11-13:
Hello I have recently upgraded to this mainboard and screen however I have been running into issues with the error "No printer attached!" whenever the printer is powered on. I am using the Marlin 2.0 Bugfix on the SKR board and I've recently updated to "TFT35_V3.0.26.x Sep 12 2020" (since this update the only way for my screen to turn on I need EXP1 and EXP2 conncected, otherwise I get a white screen but I can still hear the taping sound of the board). I've checked through various other sources and found a list of common problems so to save time I will confirm:

Within the configuration.h of my board serial ports -1 and 0 are defined at a baudrate of 115200 (in accordance with the screen settings, yes I am 100% sure this is correct), the smart controller is defined properly and all others are disabled:

#define SERIAL_PORT -1

#define SERIAL_PORT_2 0

#define BAUDRATE 115200

#define REPRAP_DISCOUNT_FULL_GRAPHIC_SMART_CONTROLLER

In addition, I believe my wiring is correct, I have read that there is a lot of misconception that the pins for the SKR 1.4 and SKR 1.4 Turbo are swapped, however I currently have it so that the single connector is innermost to the board. Thank you to for any help!

oldman4U, 2020-11-13:
Compile your own firmware from the latest master branch following the description which you can find on the main screen of the tft firmware repository. There you also find more information.

BlackMalakDown, 2020-11-14:
> Compile your own firmware from the latest master branch following the description which you can find on the main screen of the tft firmware repository. There you also find more information.

Already did this, as mentioned it is currently at firmware version "TFT35_V3.0.26.x Sep 12 2020"

oldman4U, 2020-11-14:
The last firmware is from this week.

oldman4U, 2020-11-14:
Exactly 4 days old.

BlackMalakDown, 2020-11-14:
> Compile your own firmware from the latest master branch following the description which you can find on the main screen of the tft firmware repository. There you also find more information.

Hm. I am downloading the current master branch, however when it updates it says "Sept 12"

oldman4U, 2020-11-15:
Self compiled firmware will show the day you compiled it. So you use pre compiled firmware or ...?

Send a picture of your cabling.

oldman4U, 2020-11-15:
[image]

Somehow mine shows another date.

oldman4U, 2020-11-18:
You get Sep. 12 because you are not using self compiled firmware but old pre compiled.

oldman4U, 2020-11-20:
It would be nice, if you could reply to your own ticket. Ignoring people trying to help you is strange.

Please try the firmware from here: https://github.com/radek8/BIGTREETECH-TouchScreenFirmware/tree/updateBIN

and let me know.

oldman4U, 2020-11-24:
Please help the community and close the ticket once you do not need it anymore. Thank you

Snakn1, 2021-01-16:
I am not sure why the other person did not response, but thanks This fixed my issue.

oldman4U, 2021-01-16:
Because there are many users simply not able to behave responsible.

Happy printing

introvertido, 2021-02-05:
Guys i have BIQU B1 printer, and right now when i try to use  the marlin option, i receive the LCD12864 simulator, and don't me allow to use this module, when i press the knob button, to enter on the touch, i receive the info printer not attached, can anyone help me on this, i am nooby, with this

closed 2022-06-07 by BlackMalakDown