bigtreetech/BIGTREETECH-TFT35-V3.0 #97: TFT35 causes hotend and bed temps to fluctuate.

adelyser, 2020-12-03:
I have the latest 3.0.26.x firmware installed, and I have connected it via TFT serial only to my SKR v1.3 on the TFT header. when I do, after about 30 seconds, the hotend and bed temps start to fluctutate +/- 4 degrees, both in sync. This happens while at idle or when up to temp, it doesn't make a difference. I have narrowed it down so when I unplug the TFT, the bed temps and hotend remain steady and solid (monitoring through Octoprint, no other displays connected). I see the same issue if I connect it via EXP1 and EXP2 as well.

oldman4U, 2020-12-03:
Disconnect Octoprint and see if this solves the issue as well.

adelyser, 2020-12-03:
Disconnecting octoprint and the Raspberry Pi doesn't change anything. I just tested that again with the USB cable disconnected for the RPI, and my temps start to fluctuate.

I now have the display connected via the TFT connection with a separate 5V power supply driving the TFT only, and my temps are once again stable. I am able to operate the printer as the serial connection for the TFT is connected and operational.

oldman4U, 2020-12-04:
Sounds like the TFT does not get enough power. I would reduce hardware
until it works. Also a bad power supply and em interference has to be
considered as a possible reason for the problem.

adelyser, 2020-12-04:
I've tried 2 different mainboards, a MKS Gen v1.4 and a SKR v1.3, both exhibit the same issue. I have a 500W 24V power supply to the board, and the voltage is steady and set spot on. I disconnected everything from the mainboard, except the thermistors and the display, and still have the issue. I also swapped in a Meanwell 24V powersupply, and still no difference.

I will try a shorter, different cable for the display to see if that helps, but that's about the only other thing I can think of other than maybe there's a bad filter component on the display causing noise on the power lines.

oldman4U, 2020-12-04:
Broken hardware is always a possible reason. Of course.

adelyser, 2020-12-14:
I have tried different cables, with and without Octoprint, but it all points back to the TFT causing the issue. I have it powered via a separate 5V supply off the power supply instead off from the 5V rail of the SKR, and it has been working fine. I suspect a hardware issue if others don't see this problem, not sure how to address that though.

oldman4U, 2020-12-14:
Would be interesting if changing the TFT solves the problem.

adelyser, 2020-12-14:
I'd be curious as well, but I don't have another to try.

oldman4U, 2020-12-14:
Maybe Santa can bring you a spare part!?

radek8, 2021-01-15:
I use this and it has no problem

[image]

https://www.aliexpress.com/item/4000500206294.html?spm=a2g0o.productlist.0.0.67b77213CoMw9R&algo_pvid=8f210edc-e34e-4fb9-920b-effa66c7ff93&algo_expid=8f210edc-e34e-4fb9-920b-effa66c7ff93-47&btsid=0b0a050b16107354151555970ed3f5&ws_ab_test=searchweb0_0,searchweb201602_,searchweb201603_

adelyser, 2021-02-20:
Turns out I just had to replace the power supply in the printer. After doing so, I tried powering the TFT from the SKR again and it seems the issue have disappeared. Must have been a filtering issue in the power supply, which was difficult to see.

closed 2021-02-20 by adelyser