bigtreetech/BIGTREETECH-TFT35-V3.0 #100: Random beeps

To0wnn, 2020-12-09:
Hi,

The board gives random beeps when installed inside my Ender 5 plus case.
BTT mode you do not see anything but in marlin mode i see it is also really doing something.
I see that it is puching the dail knob somehow.
When i remove the tft out of the case no beeps/pushes occur any more.
It also only happens when the bed/nozzel heating is on and not started any print or what so ever.
Bed heating is on a separate mosfet.
Also used two different screen and boards same issue.

It is on a SKR PRO V1.1 board running the latest marlin.
Looks like interference but no idea how to solve it (except for removing it from the case of course)

oldman4U, 2020-12-09:
Insulate the case so that the metal does not touch the TFT and do not tighten the screws to the max.

To0wnn, 2020-12-14:
It is sitting in a PLA printed case and no srews at al still the same somehow.
Can it be that the SKR pro board is to close to the metal case? EXP1 and 2 are close to it.

oldman4U, 2020-12-14:
You mean that electromagnetic interference causes this?

Possible. But I would start with shielded cables and check the mainboard
after that.

To0wnn, 2020-12-14:
Yes meayby i also do know otherwise..

Shielded ribbon cables?

Here are some pictures of my setup. When it is like this it looks like the knob is pressed.

[image]
[image]

oldman4U, 2020-12-15:
You can use any kind of cable for a test. You could also try to remove the ground for a test.

oldman4U, 2020-12-16:
Which TFT firmware are you using?

To0wnn, 2020-12-17:
Removing ground also no luck..

> Which TFT firmware are you using?

Latest.

oldman4U, 2020-12-17:
There is no date and time but latest written on it?!

conorlmcbride, 2021-01-13:
I’m also getting random clicks and beeps, often it just sends the menu back to the mode selection screen. This is very frustrating. Installed in the stock ender3 case it was designed for with the cables that came with the TFT... is there an update from BTT?

oldman4U, 2021-01-13:
See if removing the TFT from the metal case helps. 

Install master firmware from today from the TFT firmware repository.

To0wnn, 2021-01-14:
Will update as well see if it helps :) what are the changes?

To0wnn, 2021-01-15:
I updated the TFT firmware without succes same issue still

radek8, 2021-01-15:
Read this
https://github.com/bigtreetech/BIGTREETECH-TouchScreenFirmware/issues/1324

radek8, 2021-01-15:
Use a shielded cable for serial communication

[image]

To0wnn, 2021-01-16:
> Use a shielded cable for serial communication
> 
> [image]

Is that shielded? Do you have an spec or where did you get the cable from? Can I maybe use a piece of Cat5e utp? Or must it be stp then?

radek8, 2021-01-16:
The UTP cable is not shielded. The FTP cable is shielded, but it is unnecessarily strong.
I used a USB cable from an old mouse.

[image]

To0wnn, 2021-01-18:
Made the same cable same issue.. Read the other issue and the only thing is to isolate screen with a ziplock bag and layers of aluminum foil. Will try this later
[image]

radek8, 2021-01-18:
A heated bed powered by PWM is a huge source of interference. If the display is within range of electromagnetic waves, it may be affected (disturbed).

To0wnn, 2021-01-19:
Tried with the foil still no luck.. removed the screen the issue is a bit less wile heating but when i put the top cap over the the tft so not in the case, full chaos on the screen haha letters and characters all over the place.

radek8, 2021-01-19:
You can still turn off the PWM bed control in Marlin

To0wnn, 2021-01-19:
Already disabled during al the tests 
[image]

radek8, 2021-01-19:
This is already ghost hunting.
Try to separate the 5V power supply from the printer to the display.
Use a separate 5V power supply (such as a USB phone charger.) To power the display. There was once a problem where the display did not work well until it had a separate power supply.

To0wnn, 2021-01-19:
Ghost hunt for sure!!! haha
I will give it a try later 
Thanks for helping/thinking with me!!