bigtreetech/BIGTREETECH-TFT35-V3.0 #102: Tft 3.5 v3.0 says no printer attached!

jayala1213, 2020-12-20:
I've already downloaded correct files to the screen and updated it. Why else do I do?
Im using a skr 1.4 turbo board. 

[image]

[image]

oldman4U, 2020-12-20:
Hi. The error means, that the TFT can not see the mainboard through via the 5pin serial cable. This can have the following reasons:

1. the mainboard is broken
2. the TFT is broken
3. the cable is broken
4. the cable is connected wrong
5. the wrong serial connection is defined in Marlin firmware (in this case it should be 0 i believe)
6. different baud rates are defined for serial communication in Marlin and the TFT firmware (you can easily change the baud rate in the menu of the TFT)

This has been working a month ago, what happened?
Can you navigate through the menus by pressing a button on the TFT?

jayala1213, 2020-12-20:
> Hi. The error means, that the TFT can not see the mainboard through via the 5pin serial cable. This can have the following reasons:
>
>
>
> 1. the mainboard is broken
>
> 2. the TFT is broken
>
> 3. the cable [...]

So yes it was working a while ago but recently I tried to connect a thermocouple amplifier to the skr board using a breadboard but it started telling me that there was no printer attached. The baud rated are the same and if tried every serial port I can. Do you think I shorted a connection when I tried to connect my breadboard to the skr?

jayala1213, 2020-12-20:
> So yes it was working a while ago but recently I tried to connect a thermocouple amplifier to the skr board using a breadboard but it started telling me that there was no printer attached. The baud [...]

Also yes, I can navigate through the menus.

oldman4U, 2020-12-21:
In case you change everything back to the status before you added the extra hardware and it still does not work - I would guess the hardware is broken. But it could also be software related so make sure you change back to defaults before declaring the hardware to be broken.

jayala1213, 2020-12-21:
> In case you change everything back to the status before you added the extra hardware and it still does not work - I would guess the hardware is broken. But it could also be software related so make sure you change back to defaults before declaring the hardware to be broken. 

Alright that makes sense, thank you. By the way, do you have any experience with the adafruit thermocouple amplifier? Or connecting thermocouples to the board?

oldman4U, 2020-12-21:
No. I use a SKR PRO for bigger heated bed‘s.

jayala1213, 2020-12-22:
> No. I use a SKR PRO for bigger heated bed‘s.
> 
> 
> > In case you change everything back to the status before you added the
> > extra hardware and it still does not work - I would guess the hardware is
> > broken. But it could also be software related so make sure you change back
> > to defaults before declaring the hardware to be broken.
> >
> > Alright that makes sense, thank you. By the way, do you have any
> > experience with the adafruit thermocouple amplifier? Or connecting
> > thermocouples to the board?

Ok no worries. Thanks for your help.

closed 2021-01-04 by jayala1213