bigtreetech/BIGTREETECH-TFT35-V3.0 #105: Screen shows unknown command "image of square box"

RetArsonFD, 2020-12-23:
I couldn't figure out how to put a pic of a square box in the text, so I had to spell it it out...but it's in the " marks on the screen.  What does this mean ??

oldman4U, 2020-12-23:
Hi.

Maybe you can attach a picture or a movie which shows the issue?

RetArsonFD, 2020-12-25:
> Hi.
> 
> Maybe you can attach a picture or a movie which shows the issue?

Thank you for replying, and attempting to help me out.  However, I figured out what the error was.  Somehow I had a limit switch unplugged.  FWIW, this board is pretty intimidating to a new person.  It has WAAAAY more functions than I know what to do with, and I probably haven't even found all of them yet.  lol.

closed 2020-12-30 by RetArsonFD