bigtreetech/BIGTREETECH-TFT35-V3.0 #107: "Print From Media"

Jhawk255, 2021-01-02:
Since I installed a new SKR mini 3 v2 and a TFT35-3 V3.0, my "print from media" option is missing from the LCD12864 Simulator. I can print from the touch screen but not the simulator. I put my old display board back on and it is not there also. It used to work just fine. Im suspecting a problem with the SKR Mini 3. anyone else have this issue ?

oldman4U, 2021-01-03:
Hi

You have to enable sdprint in Marlin which is off by default.

Jhawk255, 2021-01-03:
Is that done from the screen or do I have to enable it in the firmware ?
Thanks !

oldman4U, 2021-01-03:
Firmware of the mainboard.

Jhawk255, 2021-01-03:
Thanks for your reply. That doesn’t make sense to disable that function by default. How are you supposed to print unless you flash the motherboard ? Anyway, I’m unable to flash my motherboard (it’s a new upgrade) and will not read the cf card. I have a replacement board coming tomorrow. Hopefully I’ll have better luck.

oldman4U, 2021-01-03:
Thats something you should ask those responsible for Marlin in the Marlin
repo.

oldman4U, 2021-01-05:
How is the replacement working?

Jhawk255, 2021-01-05:
Seems to be working fine !!! Now I just need to find the proper firmware to
install my BLTouch V3.1

Thanks

oldman4U, 2021-01-05:
Cool.

Happy printing 

Please help the community and close the ticket once you do not need it anymore. Thank you

closed 2021-01-07 by Jhawk255