bigtreetech/BIGTREETECH-TFT35-V3.0 #108: SKR 1.4 Turbo and TFT 35 V3 No printer attached

thompsonrem18, 2021-01-05:
I bought an skr 1.4 turbo bundled with the BTT TFT 35 V3 and compiled marlin 2.0 for my drivers and my ender 5 plus following the Kersey Fabrications videos and Teaching Tech's videos on the TFT 35 as well.
I receive a no printer attached error and can not use marlin or touch mode. I've tried all manners of swapping the exp connectors around. Serial Port 1 and Serial port 2 have been flipped between 0 and -1 multiple times. 
If the tft cable isn't plugged in I get no power to the screen. The screen receives faint power from the tft port on the screen, will illuminate for a few seconds then fade to black, still saying no printer attached. When plugged in to the port opposite the exp ports up top with the expansion ports, the screen will remain powered on and I can switch between marlin and touch mode, but have no connection with either of them. 

I can connect to the printer via octoprint on the mainsboard just fine but having a working display would be ideal. Any suggestions are very welcome!

oldman4U, 2021-01-05:
Have you read the ReadMe on the front page?

Attach a movie which shows what you describe.

oldman4U, 2021-01-05:
And attach a picture with the cable connection.

thompsonrem18, 2021-01-09:
Attached are links to videos with the issues I described. I apologize for the delay, I haven't had much time to much about with it this week. The bottom link is the FW I have tried using most recently, which worked with almost the exact same configuration I'm using now the only difference being the drivers used, everything else is identical.

https://youtu.be/evZJCiVZoxs

https://youtu.be/3Rrasaqen_w

https://github.com/tripodsgarage/Ender-5-Plus-build/blob/master/Updated%20-%20Marlin-Cheetah%205_0%20With%20Firmware.zip

thompsonrem18, 2021-01-09:
Turns out I had the bl touch wired up wrong and that was my issue all along... BL touch wiring matters

closed 2021-01-09 by thompsonrem18

Hoza12, 2022-11-28:
i think i have same problem i'm new with skr board and i loaded firmware completely but it still writes ( no printer atached )
and display goes to blank in marlin mode can any one help pls?