bigtreetech/BIGTREETECH-TFT35-V3.0 #111: TFT35 v3.0 e Esp01 S no connessione stampante

onil80, 2021-01-17:
Buongiorno, spero mi potiate aiutare, ho configurato Esp01 S, fino a collegarmi tramite indirizzo ip tutto ok, ma quando voglio collegare la stampante non ci riesco, esiste un firmware per TFT35 già compilato per esp01?

radek8, 2021-01-19:
The wifi module connected to the TFT works normally, no settings are required on the TFT

radek8, 2021-01-19:
Please send in English

onil80, 2021-01-19:
i connected esp01 S1 to tft but it doesn't connect to printer

radek8, 2021-01-19:
do you see wifi on the network?

onil80, 2021-01-19:
esp01 connects to my wi-fi network .... I connect to the esp3d interface, but I cannot connect to the printer

radek8, 2021-01-19:
answer all the questions.

onil80, 2021-01-19:
yes the tft writes to be connected,
i used both 115200 and 250000

radek8, 2021-01-19:
What record do you have with the Marlins?

onil80, 2021-01-19:
Marlin 2.0.7.2

radek8, 2021-01-19:
I'm not asking about the Marlin version, but about the board model

onil80, 2021-01-19:
sorry, skr 1.4 turbo

radek8, 2021-01-19:
You must set the communication speed the same as the one you have set on the display

onil80, 2021-01-19:
yes i know but it doesn't work anyway

radek8, 2021-01-19:
I have the same configuration and ESP works fine

radek8, 2021-01-19:
https://youtu.be/Dj6bc1YF-wk

onil80, 2021-01-19:
I programmed the esp01 with Arduino, could this be the problem?

radek8, 2021-01-19:
Probably yes

onil80, 2021-01-19:
tomorrow I receive the form for writing and repeating

radek8, 2021-01-19:
if there is a problem with the WiFi connector on the TFT, you can use the WiFi connector on the SKR V1.4.
However, there is a need to configure the port on Marlin

onil80, 2021-01-19:
I have already tried to use the wifi connector of skr1.4 and it doesn't
even work

radek8, 2021-01-19:
In order for the wifi port to work on the SKR, you must define serial port 3.
If you are using 2.0.7.2 you will need to make further adjustments.
However, I recommend using bugfix-2.0.x, further modifications will not be necessary and you will solve the problem with the encoder.

radek8, 2021-01-19:
wifi on SKR V1.4

https://youtu.be/vq6Jx1QF02o

onil80, 2021-01-19:
but if i set serial port 3, then the display doesn't work

radek8, 2021-01-19:
There are two serial ports that you can define
-1 USB port
0 TFT display connector
3 WIFI connector

If you define TFT and WIFI, .... You no longer need USB

radek8, 2021-01-19:
#define SERIAL_PORT 0
#define SERIAL_PORT_2 3

radek8, 2021-01-19:
Wifi can also be connected as a virtual serial port to a PC and use the Pronterface program

radek8, 2021-01-19:
Try WIFI for you first with a functional and verified FW and when it works, you can install your own which you can then debug.

radek8, 2021-01-19:
[FW5 funkční.zip](https://github.com/bigtreetech/BIGTREETECH-TFT35-V3.0/files/5838601/FW5.funkcni.zip)

onil80, 2021-01-20:
hi, I wanted to tell you that now everything works, I programmed the esp01 with the btt v1.0 and everything works perfectly

radek8, 2021-02-01:
Please close the ticket
Thank you