bigtreetech/BIGTREETECH-TFT35-V3.0 #112: Unknown command M702 T0 when unloading filament

Mentorman, 2021-01-21:
I get this message when using unload.
[image]

Mentorman, 2021-01-21:
I get this message when using unload.
[image]

I get the M701 version of the error when I try the load option.

I have updated the screen firmware a couple of times this month with same issue. Upgraded to an SKR 1.4 Turbo recently. I can load and unload fine with the extrude options. This error is just annoying.

oldman4U, 2021-01-21:
Please close this ticket here and open it in the TFT firmware repository after updating to the master version from today. 

This is software and not hardware related.

radek8, 2021-01-22:
Looks like you're not allowed in Marlin
#define ADVANCED_PAUSE_FEATURE

emklap, 2021-02-07:
I had the same, I had the uload/load and preheat icons but still a M702 error.
I recompiled the Marlin Marlin-bugfix-2.0.x-SKR-V1.4 branch and double-checked in  Visual Studio Code that #define ADVANCED_PAUSE_FEATURE and #define FILAMENT_LOAD_UNLOAD_GCODES were defined and in gcode.cpp the lines for the gCode 701 and 702 were active and compiled. After reloading Marlin the error was gone.