bigtreetech/BIGTREETECH-TFT35-V3.0 #113: No control once installed.  Display is on but no control.

steveandpeggyb, 2021-01-26:
I am trying to install the new (to me) TFT35-E3 v3.0 display.  The display comes on, and I have the two options of Marlin mode or Touch mode.  However, it does not recognize my touch or the control knob.  I cannot get anything to work.  If I put my old factory display on, things seem to work.  I have the BTT SKR Mini V2.0 install and now installing the TFT touch screen.  

I've swapped three different ribbon cables to no avail.

Any thoughts?

Thanks,
Steve

oldman4U, 2021-01-26:
Wrong repository for the E3 TFT.

Read the readme on the front page of the firmware repository for how to reset the TFT.

steveandpeggyb, 2021-01-27:
Sorry, missed that but solved the problem with nylon washers.

closed 2021-01-27 by steveandpeggyb