bigtreetech/BIGTREETECH-TFT35-V3.0 #118: Details on leveling.c, 5-point bed leveling positioning

Dorkblaster, 2021-02-23:
The 5 point movement feature (center and 4 corners) doesn't properly read the bed size. Seems to use the dimensions for an Ender 3 size machine. I have a larger bed (Ender 5 Plus) - ABL works fine.

oldman4U, 2021-02-25:
Hi

If this is really a problem it is in software and therefore the wrong repository.

radek8, 2021-02-27:
You must define the bed size correctly in the Config.ini file

Dorkblaster, 2021-02-28:
I did modify the config.ini for the bed size, yet the behavior didn't change (5 points for a smaller bed) instead of the larger 350mm x 350mm of the Ender Plus. I will keep looking, but for now I just wrote some gcode to do this instead. Thanks for the replies all.

closed 2021-02-28 by Dorkblaster

radek8, 2021-02-28:
Option "level_edge_distance:" 
Sets the distance of 4 points from the edge of the bed.