bigtreetech/BIGTREETECH-TFT35-V3.0 #119: TFT-35 E3 V3.0 goes blank after 5 seconds of turn on of an Ender 3 pro with BIGTREETECH SKR Mini E3 V2.0

cyburrs, 2021-03-02:
This is the same issue I read about in closed ticket #75 but there was no resolution for being able to add the BL Touch back into the system. Any new information available?

TFT-35 E3 V3.0 goes blank after 5 seconds of turn on of an Ender 3 pro with BIGTREETECH SKR Mini E3 V2.0 #75

https://github.com/bigtreetech/BIGTREETECH-TFT35-V3.0/issues/75

I have the exact same problem. I bought both the TFT 3.5 V3 and the SKR Mini E3 V2.0 just within the last two weeks.
Updated the firmware to Ender 5 with BLTouch.
Quits after 5 seconds.
Removed the BLT touch and it works. Now how do I get my BL Touch to work?

radek8, 2021-03-03:
From a comparison scheme with wiring connectors, it appears that Bigtreetech has a problem with the documentation or wiring
Z-PROBE connector
the diagram is:
pin1 - PC14
pin2 - GND

in PIN-OUT connection:
pin1 - GND
pin2 - PC14

I would try to swap these pins on the connector

[image]

[image]

radek8, 2021-03-03:
I am sorry. I anticipated the involvement

cyburrs, 2021-03-04:
Any idea about the V2.0 board? Is it the same?

blkopler, 2021-04-14:
I had the same problem, but after a while I discovered that the connector was wrong from BLTouch.

The wires, in my case, came out of the BLTouch as White/Black/Yellow/Red/Blue all down to the connector, but the connector suddenly switched Red and Blue so the connector was White/Black and Yellow/Blue/Red. So I had to stick some Male to Female Dupont Wires into the connector und right this mistake.

So please make sure that the wires are the right order.