bigtreetech/BIGTREETECH-TFT35-V3.0 #120: "No printer attached!" Error with btt SKR 1.4 Turbo - rerun

whiney1, 2021-03-04:
Hi - I've got a problem trying to connect to this TFT that I can't fix. 

**My setup:**

- BTT SKR V1.4 Turbo
- Marlin 2.0
- BTT TFT35 V3.0

**Symptoms:**

- Marlin mode works
- Touchscreen mode does not work 'No printer attached!' - No hotend data, no commands work, etc.

**Things I've tried:**

- Ensuring SERIAL_PORT(s) and baud rate are set in Marlin. I've tried 0/-1, -1/0. Baud is 115200
- Ensuring serial on TFT is 115200
- Ensuring SDSUPPORT is defined
- Trying 250000 and 57600 baud rate (matching Marlin and TFT each time)
- checking cable continuity from TFT board pins to SKR pins (no issues)
- Checking continuity on TFT pins between themselves (there is another issue that said there is a dodgy batch of TFT with internal shorts)
- switching Rx/Tx wires to see if there was a mismatch
- Compiling my own TFT firmware and flashing it (source downloaded today)
- Checking notes in TFT config.ini to make sure Marlin setting align

**Weird behaviour:**
I use Repetier normally. When I have everything configured properly, and I connect with Repetier on my PC, the TFT starts continuously beeping. You can see during this beeping that the hotend temperature actually displays on the screen but it is unresponsive until you reset it. Status shows an error 'echo:' or similar but it flashes so is hard to read and seems to be cut off.
Interestingly, when I connect with Pronterface, there are no errors.

This makes me think that the TFT/board is physically configured correctly, but that there is some issue with the serial ports configuration. But it has me stumped. Any ideas?

TIA

whiney1, 2021-03-04:
If it matters, I am running a delta printer with TMC2209 drivers and sensorless homing, and a knock-off BLTouch.

radek8, 2021-03-04:
If you send a command from Pronterface
M503
M115
M114
Does TFT connect?

radek8, 2021-03-04:
send Marlin and TFT configuration files

whiney1, 2021-03-05:
Nope, still says 'No printer attached!'. 
Interestingly, M115 only works about half the time. Sometimes it just stays on 'SENDING:M115' with no response, send it again and it outputs properly. 

Appreciate you taking the time to answer!

[Marlin and TFT config files.zip](https://github.com/bigtreetech/BIGTREETECH-TFT35-V3.0/files/6087275/Marlin.and.TFT.config.files.zip)

radek8, 2021-03-05:
You have different serial port speeds in the included configuration files.

radek8, 2021-03-05:
It looks like a faulty serial port pin on the SKR or TFT connector.
Want to try another port?
It has happened several times that the RS232 connector on the TFT was defective

DevilsDJ1990, 2021-03-07:
Hi I am using the SKR 1.4 Turbo and TFT.35 E3 V3 and Im having the same issue. I am currently making a custom machine of my design and was wondering a lot about what board to use - Lerdge K or this one and decided on this one. 
I am a total noob at programming , this was my first attempt at Marlin , so far I've only been using the stock firmware's on my printers.
So far I've tryed serial ports 0/-1 , -1/0 , baudrate 115200 , 250000 making sure its the same on both TFT and board. I installed a few different versions on the TFT and I think I am using the latest one now , also using marlin 2.07
The last thin after reading this post was trying what @radek8 said about disconnecting the RX and TX cables from the RS323 and connecting them on the UART3 and uploading the modified firmware he provided and still nothing.
Forgot to mention in Marlin mode everything works (now after 10 hours ) but still No Printer Attached on TFT mode.
Im attaching my config files for board and TFT(or at least I think this was the last one I used)
[Marlin and TFT config.zip](https://github.com/bigtreetech/BIGTREETECH-TFT35-V3.0/files/6097652/Marlin.and.TFT.config.zip)

DevilsDJ1990, 2021-03-07:
Here is how I have connected the TFT to the board as @radek8 mentioned 
[image]
[image]

DevilsDJ1990, 2021-03-07:
Just now downloaded this [https://github.com/radek8/BIGTREETECH-TouchScreenFirmware](https://github.com/radek8/BIGTREETECH-TouchScreenFirmware) , and put from folder "Copy to SD CARD" - config.ini , BIGTREE_TFT35_V3.0_E3.26.x.bin , language_en.ini , and the TFT35 folder from THEME_Unified Menu Material theme to and SD card and inserted it on the TFT waited to do it's thing , put cables and original pin locations and still nothing.

radek8, 2021-03-07:
You have it badly connected

[image]

radek8, 2021-03-07:
There is often a problem with the serial cable.
First of all, measure if the cable is OK.

DevilsDJ1990, 2021-03-07:
@radek8  Yeah I noticed , but after I updated with  https://github.com/radek8/BIGTREETECH-TouchScreenFirmware I cant use the bin file you posted here , can you possible modify one for the E3 version for me as so far I have no idea on how to compile one in VSCode for the TFT , barely managed to compile one for MArlin for the board. Also tried 2 different cables , the one you see in the photo and the stock one that came with the board and tft

radek8, 2021-03-07:
[BIGTREE_TFT35_V3.0_E3.26.x.zip](https://github.com/bigtreetech/BIGTREETECH-TFT35-V3.0/files/6097826/BIGTREE_TFT35_V3.0_E3.26.x.zip)

DevilsDJ1990, 2021-03-07:
@radek8 Thank you alot , so now I do this right and just put all of these on an SD card and into the TFT
[image]

radek8, 2021-03-07:
Yes

radek8, 2021-03-07:
Be careful you have the old date on other files.
Instead, download the current one.

DevilsDJ1990, 2021-03-07:
@radek8 got some errors for language and icons , which would be the current ones ? So many different places with files im getting confused alot

radek8, 2021-03-07:
Which icon set do you prefer?

DevilsDJ1990, 2021-03-07:
@radek8 Do I download this one ? [https://github.com/bigtreetech/BIGTREETECH-TouchScreenFirmware](https://github.com/bigtreetech/BIGTREETECH-TouchScreenFirmware)  and use the language , config , TFT35 folder from there and your bin file ?

radek8, 2021-03-07:
the language pack is here

https://github.com/bigtreetech/BIGTREETECH-TouchScreenFirmware/tree/master/Copy%20to%20SD%20Card%20root%20directory%20to%20update/Language%20Packs

the config.ini is here

https://github.com/bigtreetech/BIGTREETECH-TouchScreenFirmware/tree/master/Copy%20to%20SD%20Card%20root%20directory%20to%20update

folder with icons
https://github.com/bigtreetech/BIGTREETECH-TouchScreenFirmware/tree/master/Copy%20to%20SD%20Card%20root%20directory%20to%20update/THEME_The%20Round%20Miracle%20Menu%20Material%20theme

bin file I sent

DevilsDJ1990, 2021-03-07:
@radek8 DUDE I LOVE YOU , it worked , FINALLY IT WORKED . no more No Printer Attached.
But just so I know for the future - so this means my TFT original pins are faulty and also did you modify something like this somewhere - not sure what it is and where and how just saw it in another post 
/* 
* SERIAL_PORT: communicating with host(Marlin, smoothieware, etc...)
 * SERIAL_PORT_X: communicating with other controller(Octoprint, ESP3D, other UART Touch Screen, etc...)
 */
#define SERIAL_PORT   _USART3  //default usart port
#define SERIAL_PORT_2 _USART1
#define SERIAL_PORT_3 _USART2
#define SERIAL_PORT_4 _UART4

radek8, 2021-03-07:
Looks like you have faulty pins on the SR232 connector.
for you, this means that you cannot use a precompiled BIN file and you have to compile your own after editing.
Editing is easy.
The red line is the original, the green one replaces it

[image]

radek8, 2021-03-07:
If you use an additional module of the 5V power supply, then you have the wrong jupmer installed.

[image]

DevilsDJ1990, 2021-03-07:
Yeah havent reached that far yet , I will be adding lights , 3D Touch sensor , might add the UPS module and the automatic power shutdown module . This is what I have so far , I have a 1000W laser at work and I work with stainless steell , I decided I wanted as little plastic parts as possible , so everything except the hotend mounts for fans is metal so far , modeled and tested by me. Including the bed support frame. Size for now is X310 , Y310 , Z650
[image]
[image]
[image]
[image]
[image]
[image]
[image]

radek8, 2021-03-07:
The X-axis belt tensioner is not good :-)

DevilsDJ1990, 2021-03-07:
I have actually ordered those , they are arriving tommorow [https://www.aliexpress.com/item/4000127067311.html?spm=a2g0s.9042311.0.0.51194c4dfePSvD](https://www.aliexpress.com/item/4000127067311.html?spm=a2g0s.9042311.0.0.51194c4dfePSvD)   , there is alot more to be done - I have to decide on how to reinforce the Z axis , something like they did on the CR10 , I have to decide where the electronics will be ( either under the bed - display in front or to make a separate enclosure for the electronics ) , I have to make mounts for cable chains for the cables for the bed and Z axis , decide on what extruder to use ( I have tryed BMG and Titan clones on other printers I own) and probably more that I cant think of now. IF you are interested I can keep you up to date somewhere and I am open to suggestions and opinions. When everything is up and working I will probably post files for the parts I made somewhere , dxf files for the metal version and stl for printed plastic ones ( since I started with test ones before making the metal version).
I was thinking maybe getting one of these [https://www.aliexpress.com/item/1005001398845300.html?spm=a2g0s.8937460.0.0.24fc2e0eaigbfI](https://www.aliexpress.com/item/1005001398845300.html?spm=a2g0s.8937460.0.0.24fc2e0eaigbfI) or making again out of metal a version for the cr10 type hotend maybe one of these for extruder but not sure if they are good [https://www.aliexpress.com/item/4000648526348.html?spm=a2g0s.8937460.0.0.1f212e0e6p9AdE](https://www.aliexpress.com/item/4000648526348.html?spm=a2g0s.8937460.0.0.1f212e0e6p9AdE) , also as I said about electronics maybe I will make them like the WANHAO

DevilsDJ1990, 2021-03-08:
@radek8 Is it here that I need to do this ? And then just compile using the newest files ?
[image]

radek8, 2021-03-08:
Yes

whiney1, 2021-03-08:
> 
> 
> It looks like a faulty serial port pin on the SKR or TFT connector.
> Want to try another port?
> It has happened several times that the RS232 connector on the TFT was defective
> 
> To test, disconnect the RX and TX pin of the SR232 cable to the TFT and connect them to the RX and TX pins on the UART3 connector.
> To do this, you need to upload the modified FW to the TFT
> [BIGTREE_TFT35_V3.0.26.x.zip](https://github.com/bigtreetech/BIGTREETECH-TFT35-V3.0/files/6093274/BIGTREE_TFT35_V3.0.26.x.zip)

You absolute legend radek8, that fixed it right up. I modified the pin file and moved the cables - the back of my TFT looks a little different than the picture that was put up in this thread but I made sure to use the Tx3/Rx3 pins.

I'm Devilsdj1990 asked you the follow up questions on how to modify the files/cables though because I would have had to do that too :D

Thanks!

closed 2021-03-08 by whiney1

Korpz58, 2023-02-11:
I apologize for this, but I am so new to this stuff I make a first grader look like a College graduate. Not sure if this is where I can post but I'm going to try. I also bought a ender 3, read all the things on the skr mini e3 v3, and the tft 35. Was able to put all the stuff in and now get the "no printer attached". Even before that I only managed to print 1 straight line. No idea how to flash firmware (although read up on it and does nothing at all. I'm ready to throw this thing through the window and it's a shame as I can do nothing at all with this. Any help would be greatly appreciated, and I truly mean that.
Thank you