bigtreetech/BIGTREETECH-TFT35-V3.0 #121: screen switching issue and mairlin issue

tree124, 2021-03-12:
when i resived my tft35 screen for my ender 3 pro the mairlin mode it would just be blank screen even through i had conected the right cables to the right place. Then after that it stopped selecting the two differnt modes this did not happen straight away but after trying to fix the first issue i discovered this one sudderly i would hold the knob down and it would just beep and not come up with the two modes for marlin and touch mode please help me.

oldman4U, 2021-03-13:
Have you updated the firmware?

tree124, 2021-03-13:
Yes I have indeed

oldman4U, 2021-03-13:
Most likely you used the wrong firmware because very likely this is the
wrong repository for your TFT.

Send a picture of the TFT and the firmware you used for updating.

tree124, 2021-03-13:
> I used the touch screen firmware for the screen and the touch screen
> firmware and original for the baird

I meant board

tree124, 2021-03-13:
[Toby's Original firmware (3).zip](https://github.com/bigtreetech/BIGTREETECH-TFT35-V3.0/files/6135627/Toby.s.Original.firmware.3.zip)
[Toby's Touch Screen firmware (3).zip](https://github.com/bigtreetech/BIGTREETECH-TFT35-V3.0/files/6135628/Toby.s.Touch.Screen.firmware.3.zip)
these are the rwo files i used for my baord

tree124, 2021-03-13:
the file that i used for the screen i got off bigtreetech gethub page

oldman4U, 2021-03-13:
Please send a picture of your TFT.

tree124, 2021-03-13:
with it plugged in and turrned on or just a normal picture of it

oldman4U, 2021-03-13:
Normal picture

And also a picture of the content of the SD card showing what you have
updated. You can take the picture from the computer screen.

tree124, 2021-03-13:
i deleated the files left on the sd card unfortunly but i did update the touch mode properly to the new style it looked really nice i will try to send a normal picture

oldman4U, 2021-03-13:
And a picture of the TFT?

tree124, 2021-03-13:
it wont let me send them because the files are to big when i took a picture of the screen

oldman4U, 2021-03-14:
The information you provided so far is not enough to help.

I guess your problem is related to the wrong firmware. Make sure to upload the correct firmware for your TFT.

tree124, 2021-03-14:
Okay sorry for the late reply been busy it is really anoining I can not
show you pictures

radek8, 2021-03-15:
> Okay sorry for the late reply been busy it is really anoining I can not show you pictures
> […](#)

???

tree124, 2021-03-15:
About the tft35 issue

radek8, 2021-03-15:
The description of your problem is not clear, you cannot send the photo, you did not send the configuration. (nothing to read from bin file)
It's hard to help you ...

tree124, 2021-03-15:
The config file is to big for the website and email I think but will try
the email again

radek8, 2021-03-16:
???
config.ini. configuration. h and configuration_adv.h are several kilobytes

tree124, 2021-03-16:
Yeah they are thee files that are to big