bigtreetech/BIGTREETECH-TFT35-V3.0 #122: SKR Display V3 Found outdated data: Icons: Update Required

XeroDay99, 2021-03-18:
SKR Display V3 

Found outdated data:

Icons: Update Required

radek8, 2021-03-19:
So update the icons.
What is the problem?

radek8, 2021-06-04:
please close this ticket

XeroDay99, 2021-06-04:
Great answer! So professional. Now I wonder people just return the product!

closed 2021-06-04 by XeroDay99

radek8, 2021-06-04:
What is unprofessional about my answers?

The error message was clear and there was no reason to write an error message.
Or am I wrong?

radek8, 2021-06-04:
I'm a 3D printer user just like you.
I try to help others.
People write requests for help here and then don't even answer their own ticket.
The fact that they should lock him up when they don't need him doesn't occur to them.
Thank you for your answer even though it was a bit sarcastic, and for closing the ticket.

PS:
It's a great product, it would be a shame to return it :-D