bigtreetech/BIGTREETECH-TFT35-V3.0 #124: Custom Firmware Binary and config.ini

jerryfudd, 2021-03-22:
Hi,

Just got the TFT35 E3 V3.0 and since I've been customising my Marlin firmware for my SKR Mini 1.2 was loving the fact I can do the same for the TFT35.

I've generated my BIGTREE_TFT35_V3.0_E3.26.x.bin and placed on the SD card along with the contents of the TFT35 with whatever menu system I want.

However, what I don't get is that on boot up the sd card firmware contents don't get applied to the screen. That is unless I include the config.ini file - which then appears to override most of my BIGTREE_TFT35_V3.0_E3.26.x.bin file.

Can the BIGTREE_TFT35_V3.0_E3.26.x.bin be applied without config.ini as at the moment it seems an annoyance that I have to set my preferences in two places - or am I doing something wrong?

jerryfudd, 2021-03-22:
issue was that I wasn't reseting to default settings after applying the .bin which now appears to be working as a solo file on the card.

oldman4U, 2021-03-25:
Put the precompiled firmware, the TFT folder of your screen and the edited config.ini on the SD card and all will be fine. Like described in the Readme of the firmware repository.

jerryfudd, 2021-03-26:
> Put the precompiled firmware, the TFT folder of your screen and the edited config.ini on the SD card and all will be fine. Like described in the Readme of the firmware repository.

all sorted, as per my reply.

oldman4U, 2021-03-26:
Perfect. There is no need to edit configuration.h or compile firmware is
what I wanted to add. Otherwise other users believe they have to what you
wrote.

Please help the community and close the ticket once you do not need it
anymore. Thank you

closed 2021-03-27 by jerryfudd