bigtreetech/BIGTREETECH-TFT35-V3.0 #125: Connecting TFT35 to a Creality Upgrade Ender 3 Pro V4.2.7 Silent Motherboard with TMC2225 Drive

AMPens, 2021-03-31:
1) where do i plug the rs232 cable from the tft on to the motherboard and which way round

2) tried to do the firmware update 
files on SD card are

BIGTREE_TFT35_V3.0_E3.26.x.bin
language_uk.ini
TFT35  ("D:\THEME_Unified Menu Material theme")
config.ini

And get an error "found outdated Data"

Now i cannot use this in either mode

radek8, 2021-03-31:
Question 1, read here
https://github.com/bigtreetech/BIGTREETECH-TouchScreenFirmware/issues/1780

Question 2 - when updating the display, follow the manual here
https://github.com/bigtreetech/BIGTREETECH-TouchScreenFirmware

radek8, 2021-03-31:
https://crosslink.io/2020/08/21/hacking-creality-mainboards-serial-port-breakout-explained/