bigtreetech/BIGTREETECH-TFT35-V3.0 #127: Serial Connection to Printer "No printer attached"

mristau, 2021-05-02:
I bought the SKR1.4 Turbo + BTT-TFT35-V3.0 in february and used both to upgrade my Anet A8 Printer.
I had some slight problems first, but after some digging all worked flawlessly.
Now i bought parts to upgrade the A8 to an AM8 with aluminium profile body.
Yesterday i attached everything together and got that "no printer attached", fine first i had the board fresh it took some time until it connected, but now nothing happened, every some reboots it changed to "3D printer ready" but i could not control it through the tft touch display.
Switching to Marlin-Mode works and i can control the printer there.

I think at some point i might have inserted the Serial-Cable flipped, i had some strange flickering on the screen, that was gone when the cable was removed.
Also i made me an extra cable with not the dupont connector but another jst-xh as  is on the display side because the dupont would sometimes fall out.

The display lights up and seems to work, even if i don't connect ext1+2 but no connection.

I think i have damaged something in that connector on either side, but maybe somebody knows how to fix that.
I also added a picture of my custom cable, brown cable connects into rst on both the tft and skr board now.

originally i had marlin 2.0.7 and 3.0.26 on the display, i tried today upgrading, so the printer has marlin 2.0.8 now and 3.0.27 both fresh from github with my configs

[image]

radek8, 2021-05-02:
Insert an empty file named reset.txt on the card, insert the card in the TFT slot, and restart the printer.

mristau, 2021-05-03:
I tried that, it did reset everything, but still no connection, i will take a usb-serial adapter from work today and test if i not destroyed the serial port.
Is it maybe possible to use one of the other serial ports of the display if i really destroyed this one, or would i have to buy a new display then.

radek8, 2021-05-03:
Yes.
If the serial pins are damaged, it is possible to use another serial port after adjusting the FW.
Several users have already used it successfully :-)

radek8, 2021-05-03:
https://github.com/bigtreetech/BIGTREETECH-TFT35-V3.0/issues/120

https://github.com/bigtreetech/BIGTREETECH-TouchScreenFirmware/issues/715

https://github.com/bigtreetech/BIGTREETECH-TouchScreenFirmware/issues/637

mristau, 2021-05-03:
I think at some point i had the 4pin connector flipped, meaning 5v/gnd from the printer connected to rx/tx of the display, i will test them and then look into switching to one of the other serial ports probably, i don't think i'll be needing all of them later.

Thanks for the help and links

radek8, 2021-05-03:
This is a no complicated modification, but it means compiling your own modified FW, but that's not a problem.
If you make your own cable, use a shielded cable. just a cable from an old PC mouse.
The serial link can be interrupted.
https://github.com/bigtreetech/BIGTREETECH-TouchScreenFirmware/issues/1747
[image]

mristau, 2021-05-03:
so i decided to connect to the uart3 port and the printer will react to the controls over touch display and connects to the printer.
Great advice with the cable, for now i use some regular ribbon cable, but i think i have some old shielded cables in some drawer.
I just put a 3pin jst-connector for the uart and the 5pin with 5v/gnd/reset

closed 2021-05-03 by mristau