bigtreetech/BIGTREETECH-TFT35-V3.0 #128: TFT35 print slows down after Probe offset change

AhmedTheGeek, 2021-05-23:
To reproduce:

- Initiate probe offset procedure.
- Save the new probe offset.
- Start a gcode file with G29 in it.
- Once the G29 is finished the print speed slows down a lot.

The issue is reproducible every time.

radek8, 2021-05-24:
I assume you are using an encoder to change the value ...
Rotating the encoder in touch mode changes the print speed in Marlin.

Write what your HW and SW configuration is
For some configurations, the problem is fixed in the new FW.
Thank you

AhmedTheGeek, 2021-05-25:
@radek8 Yeah right the last few times did this I was using the encoder.

HW:
- Ender 3 Pro.
- SKR Mini E3 V2.
- TFT 35 E3 v3.0.

SW:
- TFT35_E3_V3.0.27.X May 17 2021 Update.
- Marlin bugfix-2.0.x Mar 18 2021 Update.

radek8, 2021-05-25:
Yes, that's your problem.
There is no fix for this SKRmini board.

There are 4 possible solutions. Choose which suits you

1. Do not use the encoder in touch mode.
2. Do not use simulated Marlin mode on TFT (disconnect 10pin cable).
3. In Marlin, turn off the return to the status screen. The speed change only takes place on the status screen. After turning on the printers, just one click on the encoder and then you can rotate the encoder as you wish. But you can't click!
4. modification of the definition of pins in Marlin and stretching of another wire between SKR and TFT. Theoretically it should work, but I haven't tried it. I don't have a SKR mini to try it out. less common but the best solution.

oldman4U, 2021-05-25:
Or buy a mainboard which connects with 2 EXP csbles😁

radek8, 2021-07-18:
If the problem is resolved, close the ticket. Thank you.

closed 2021-07-18 by AhmedTheGeek