bigtreetech/BIGTREETECH-TFT35-V3.0 #129: Problems with "No printer attached"

freytod, 2021-06-23:
Hi,
i just got my TFT35-E3-V3.0 and a SKR mini E3 V3. I want to use both on an Ender 5 (default/early 2019) with BLTouch.
The board works fine but the screen doesn't seem to find the board. 
I can use the marlin-mode without any problems but the touchscreen-mode wont work, no matter what I have tried.

Things I tried:
-changed the baudrate on TFT and marlin/configuration.h between 115200 and 250000
-changed SERIAL_PORT and SERIAL_PORT_2 between -1 and 2 in every possible configuration
-used my own compile of Marlin 2.0.9 as well as BTTs precompiled version for Ender5
-searched in other threads and found a [file](https://www.dropbox.com/s/k7tb4sck5d2h5bb/BTT%20TFT%20to%20BTT%20Mainboards%202021.pdf?dl=0) where the serial_ports settings are mentioned
-tested the serialcable and also replaced it with wires
-tried to use UART3 instead of the serialport like mentioned [here](https://github.com/bigtreetech/BIGTREETECH-TFT35-V3.0/issues/120#issuecomment-791666846) and used a custom compiled version of the FW with changes mentioned [here](https://github.com/bigtreetech/BIGTREETECH-TFT35-V3.0/issues/120#issuecomment-792350134) (pictures below)
[image]
[image]
[image]

I have no Idea what to try anymore and my best guess is, that my TFT is just broken. 
Maybe someone can help me if possible.

Thanks and best regards

[Configuration.zip](https://github.com/bigtreetech/BIGTREETECH-TFT35-V3.0/files/6704623/Configuration.zip)

radek8, 2021-06-23:
According to the photo, you have a rotated connector

freytod, 2021-06-23:
So I just have to switch the pins of the original cable on the SKR mini?

radek8, 2021-06-23:
A brown cable goes to the RESET pin on the SKR.
A blue cable goes to the RESET pin on the TFT.

radek8, 2021-06-23:
The connector is rotated 180 °

radek8, 2021-06-23:
Sorry my mistake, it's fine

radek8, 2021-06-23:
The right setting is for you    SERIAL_PORT 2

freytod, 2021-06-23:
> 
> 
> The right setting is for you SERIAL_PORT 2

I have SERIAL_PORT 2 in my config but it still doesnt want to work
But thanks for your help! I appreciate it.

radek8, 2021-06-23:
Do you have the same serial port speed on both?
115200

freytod, 2021-06-23:
> 
> 
> Do you have the same serial port speed on both?
> 115200

Yes, my latest configuration.h is attached to the initial post if that helps...

radek8, 2021-06-23:
There is no ini file for TFT.
So I don't know what the speed is on TFT

radek8, 2021-06-23:
Try commenting on a line in Marlim
#define BAUDRATE_2 115200 // Enable to override BAUDRATE
This is a new feature, there could be a problem.

freytod, 2021-06-23:
> 
> 
> There is no ini file for TFT.
> So I don't know what the speed is on TFT

You are right, my bad. Attached the ini.

> 
> 
> Try commenting on a line in Marlim
> #define BAUDRATE_2 115200 // Enable to override BAUDRATE
> This is a new feature, there could be a problem.

Tried it with and without #define BAUDRATE_2 115200
Doesn't make a difference

radek8, 2021-06-23:
If you connect the printer to a PC via USB and send the M503 S0 command from Pronterface
Will TFT respond?

freytod, 2021-06-23:
> 
> 
> If you connect the printer via USB to a PC and send the M503 S0 command from the PC from Pronterface
> Will TFT respond?

No, nothing happens if I do that

radek8, 2021-06-23:
If you have correctly tested the standard serial port on the TFT and also UART3 and both e behave the same, it is unlikely that both will be defective. Then there would be a problem with the SKR serial connector

freytod, 2021-06-23:
I hope I have correctly tested everything with the standard port and the UART3... Wasn't sure if the changes I made to the TFT FW are in pin_tft_35_v3_0.h and not in pin_tft_35_e3_v3_0.h.
Is there a difference? pin_tft_35_e3_v3_0.h is nearly empty

radek8, 2021-06-24:
It happened several times that the serial connection cable was defective. Can you try another cable between TFT and SKR?

To your question? TFT 35 V3 and TFT 35 E3 V3 both use the same set of pins pin_tft_35_v3_0.h
TFT 35 E3 V3 also the file pin_tft_35_e3_v3_0.h
In both cases, you are editing the same file.

radek8, 2021-06-24:
Do you have the option to try another TFT, or connect your TFT to another printer (another Marlin motherboard)?

freytod, 2021-06-24:
> 
> 
> It happened several times that the serial connection cable was defective. Can you try another cable between TFT and SKR?

I tried to connect the TFT by using the singlecable connections in the serialport with the FW provided by BTT. Doesnt change anything..

> 
> 
> Do you have the option to try another TFT, or connect your TFT to another printer (another Marlin motherboard)?

The only other TFT i have is the originial 12864 and the only other motherboard is the Creality V1.1.2. Not sure if this helps since connection via EXP3 works just fine.

radek8, 2021-06-24:
Neither the original display nor the original Ender board can be used to test functionality.

The SKR could theoretically test the connection to serial port 1 to rule out the possibility that there is a faulty pin on the SKR connector, similar to what you did with remapping ports to TFT.

I don't have a SKR mini E3 to try it out, but according to the wiring diagram it should work.

freytod, 2021-06-24:
> 
> 
> The SKR could theoretically test the connection to serial port 1 to rule out the possibility that there is a faulty pin on the SKR connector, similar to what you did with remapping ports to TFT.

How can I test the connection? I'm not very experienced in those things tbh..

radek8, 2021-06-24:
I'm at work now and I don't have time. In the evening, I could look at and write how and where to get involved, and how to edit Marlin.

freytod, 2021-06-24:
> 
> 
> I'm at work now and I don't have time. In the evening, I could look at and write how and where to get involved, and how to edit Marlin.

That would be a great help, thank you!

freytod, 2021-06-24:
I looked at the pin-diagramms a little bit. Maybe I'm wrong but shouldn't Tx on the SKR go to Rx on the TFT and vice versa? If I connect the serialcable, it would connect Tx to Tx and Rx to Rx... Will try to connect them manually.

radek8, 2021-06-24:
You can try both variants

radek8, 2021-06-24:
Upload the original FW to the TFT. Connect the serial connector to the RS232 socket.
Set the port speed to 115200.
Disconnect the 10Pin cable between the SKR and the TFT.
Connect the serial cable. but without the RX and TX pin on the SKR
Connect the RX and TX contacts to pins RT1 TX1 in the EXP connector.

[image]

Set in Marlin
#define SERIAL_PORT 1
#define BAUDRATE 115200

In the file
pins_BTT_SKR_MINI_E3_common.h
comment lines 146 and 147
// # define BTN_EN1 PA9
// # define BTN_EN2 PA10

Record a modified Marlin.

radek8, 2021-06-24:
Do you have these options enabled (uncommented)?

#define M115_GEOMETRY_REPORT
#define M114_DETAIL

freytod, 2021-06-24:
> 
> 
> Do you have these options enabled (uncommented)?
> 
> #define M115_GEOMETRY_REPORT
> #define M114_DETAIL

No, shall I?

If I do, what you suggested in the post above, the display stays black...

radek8, 2021-06-24:
> > Do you have these options enabled (uncommented)?
> > #define M115_GEOMETRY_REPORT
> > #define M114_DETAIL
> 
> No, shall I?

Yes you must !!!

radek8, 2021-06-24:
> 
> If I do, what you suggested in the post above, the display stays black...

Do you have switched to touch mode ?!

freytod, 2021-06-24:
> 
> 
> > If I do, what you suggested in the post above, the display stays black...
> 
> Do you have switched to touch mode ?!

yes, touchmode is on defaultä

radek8, 2021-06-24:
Touch mode must boot even if the serial port does not work.

radek8, 2021-06-24:
When is the display black?

freytod, 2021-06-24:
The  serialport doesnt seem to give power anymore. It has yesterday without doubt. Tried it again with a normal fw on the SKR. Still black. If I connect SKR-EXP1/EXP3-TFT, the TFT works.

radek8, 2021-06-24:
send photo of engagement

radek8, 2021-06-24:
It must only work with a serial 5 pin connector

freytod, 2021-06-24:
> 
> 
> It must only work with a serial 5 pin connector

It doesn't. I tried somethings and by using the 5V power of the EXP1 on the SKR, it seems to work again. 
[image]
[image]

connected it like you suggested with the firmware changes you suggested. Still "no printer attached"
I'm feeling a bit dumb right now since I c an't figure out the problem...

freytod, 2021-06-24:
Okay, I changed red/orange and I got "failed to enable bed leveling" but the printer seems to be attached now. It shows at least one temperature.

radek8, 2021-06-24:
This means that the display is OK and the problem is with the SKR on the TFT connector

freytod, 2021-06-24:
> 
> 
> This means that the display is OK and the problem is with the SKR on the TFT connector

Yes, I think so, too. I will try to exchange the board. Sadly it was a bundle, so I may get a good SKR but a bad TFT in the next package... Hopefully they accept the exchange of just one product. 

Thanks a lot for your help! Wouldn't have gotten that far without you.

radek8, 2021-06-24:
If they do not accept the complaint, it can be used in an emergency. Only the connection of the encoder would be solved.

radek8, 2021-06-24:
One more question.
Did you enable M115 and M114 only now, or did you try the original connection with the M115 and M114 option enabled?
Disabled M114 and M115 may cause a printer not connected message.

freytod, 2021-06-24:
> 
> 
> One more question.
> Did you enable M115 and M114 only now, or did you try the original connection with the M115 and M114 option enabled?
> Disabled M114 and M115 may cause a printer not connected message.

Just tried it now with the last build of my fw for the SKR. 
I will try again with the normal connection and respond.

radek8, 2021-06-24:
Don't forget to change too
#define SERIAL_PORT 2

freytod, 2021-06-24:
> 
> 
> Don't forget to change too
> #define SERIAL_PORT 2

Did it in two different folders anyways. But I still got the problem of no power over serial. I can just power it with EXP. And than, the TFT still doesn't work. So the serialport of the SKR is broken.

radek8, 2021-06-24:
Do you have the red jumper in the correct position?

freytod, 2021-06-24:
> 
> 
> Do you have the red jumper in the correct position?

Which red jumper are you talking about?

radek8, 2021-06-24:
[image]

freytod, 2021-06-24:
Isn't that one for neo pixel?

radek8, 2021-06-24:
I don't know how it is with SKRminiE3, but with SKRV1.4 it switches the power supply of the probe neopixel and TFT between the integrated 5V source and the additional 5V module (which you don't have). In the wrong position, the TFT connector would not be powered.

freytod, 2021-06-24:
Seems to be the correct way. It powers TFT and BLTouch if I activate the onboard 5V. My BLTouch has power.

radek8, 2021-06-24:
So what is the conclusion?
Faulty pins on the SKR in the TFT connector?

freytod, 2021-06-24:
My conclusion is faulty pins on the SKR. That i get no power over the serialport is just one thing. I will try to exchange the SKR alone. 
Thanks again for your help and your time.

radek8, 2021-06-24:
Have a nice day

freytod, 2021-06-24:
Thanks, you too.
I will leave this issue open until I get the new board.

radek8, 2021-07-02:
Hello
It's possible that your SKR board was OK.
The problem could be in Marin.
It is said that it does not communicate on serial port 0
see:
https://github.com/MarlinFirmware/Marlin/issues/22283
https://github.com/MarlinFirmware/Marlin/pull/22284

freytod, 2021-07-02:
> Hello
> It's possible that your SKR board was OK.
> The problem could be in Marin.
> It is said that it does not communicate on serial port 0
> see:
> [MarlinFirmware/Marlin#22283](https://github.com/MarlinFirmware/Marlin/issues/22283)
> [MarlinFirmware/Marlin#22284](https://github.com/MarlinFirmware/Marlin/pull/22284)

Thank you for that! But I measured the voltage of the serial port and it was 0. I also tested if there was a connection between the 5V-rail (the red jumper) and the 5V pin. Also nothing. The connector is faulty anyways and I will receive a new one. Took me nearly a week of sending proofs to the seller... :D

radek8, 2021-07-02:
OK
The problem is said to be at speeds of 250000 and higher, you tested at 115200.

freytod, 2021-07-04:
Got my new SKR yesterday.
Just installed the Firmware and everything worked out fine without issues. The only thing I have left is that my TFT shows "failed to enable bed leveling" on startup. I guess I just have to disable it since the BLTouch works as a Z-probe. 
Thank you for your help, I will close this thread soon.

radek8, 2021-07-27:
If you no longer have a problem, can you close the ticket?

[1 more reply]

closed 2021-07-27 by freytod