bigtreetech/BIGTREETECH-TFT35-V3.0 #130: ESP01s stopped working after upgrade firmware

belkevglaz, 2021-06-26:
I have esp01s connected to tft35 diplay. All working well before. 
After upgrade to the last commit c7842cf8f7d4bfe0395cbaba4ed538ce005af581 esp module not working  (no wifi conenctions and led blinking on the board )
How to fix it ? Need to do some configurations in firmware ? 
Thanks

oldman4U, 2021-06-26:
You should ask this in the firmware repository.

With the latest firmware version, all serial ports except the one which connects with the mainboard are disabled by default but can be turned on by editing the related part of the config.ini.

belkevglaz, 2021-06-26:
Could you point me please in config.ini where it is ?

belkevglaz, 2021-06-26:
You are talking about `multi_serial:0` ? My esp connected in EXP1 port, what I need to enable, `multi-serial:1` ?

belkevglaz, 2021-06-26:
Updated firmware with `multi_serial:7` ( all serial ports are enabled) does not help. esp still not work.

oldman4U, 2021-06-27:
Restart after firmware update. 

You have it connected to the dedicated WIFI connector.[image]

belkevglaz, 2021-06-27:
Fixed. Problem was in pins contact. Re-installed esp and it started to work.

closed 2021-06-27 by belkevglaz