bigtreetech/BIGTREETECH-TFT35-V3.0 #131: ABL Error! (SKR 1.4 + TFT35 V3.0)

samed5497, 2021-07-04:
I have been manufacturing 3D printers in Turkey for a long time.  The cards I use are generally MKS products.  I wanted to use BTT Products on my last device (you can check it on YouTube: IDEXBOT).  I have no problems with mechanics and electronics, but there is one point that I can't get over with software.

I am using 'bltouch' probe for automatic table leveling.  I am using Marlin 2.0.9 and 'ABL_Billiear' is active. I have the same problem with older versions of Marlin.  The device receives all commands, takes measurements and saves them.  However, when I start printing, the leveling of the Z axis does not work correctly.  Although automatic table leveling is active (M420 S1 command) it does not work correctly.  When I look at the mesh measurements, the distances are correct.  I think the device reverses the received points in itself.  I also set it with the mini step, but I'm having problems again in the next print.  

I have never had a problem with MKS products.  Automatic likes are constantly working incorrectly.  I need real help on this.  Is there anyone who can help?

Thank you :)

radek8, 2021-07-09:
In the Simulovs of Marlin mode, the problem of heaven?

samed5497, 2021-07-10:
Yes I am having the same problem.  I think there is a communication problem between tft35 and skr1.4.  I made all the settings as in the information file.  But this is my 3rd device with this problem.  I think their software is conflicting with each other.

radek8, 2021-07-10:
If the same problem is in both touch and Marlin mode, there is no problem with the display firmware.
The problem is in the Marlin configuration, or Marlin has an error.
read
https://github.com/MarlinFirmware/Marlin/issues/22301

samed5497, 2021-07-18:
I tried it on most versions of Marlin.  problem not solved.  While calibrating the TFT screen, I activated everything necessary in the description section.  My problem is still not fixed.  I am having the same problem as in this example.

radek8, 2021-07-18:
Can you insert an M503 and M115 listing here?

XDA-Bam, 2021-07-20:
> 
> 
> I tried it on most versions of Marlin. problem not solved. While calibrating the TFT screen, I activated everything necessary in the description section. My problem is still not fixed. I am having the same problem as in this example.

"most versions" is not very precise - please list the exact versions and also check older revisions from 2020, if they have the same problem.

This might also have been fixed by 154decf, which is why you should test the latest bugfix-2.0.x before anything else.

samed5497, 2021-07-26:
I compiled version 2.0.9 from scratch and the problem went away. I think there was a conflict as a result of the changes I made in the configuration files. Thank you anyway.

closed 2021-07-26 by samed5497