bigtreetech/BIGTREETECH-TFT35-V3.0 #132: Problem with ABL level corners

bubulgum, 2021-07-09:
hi, all.
I'm using Mega zero 2 with SKR mini E3 v2 and TFT35 E3.
Firmware KAD_marlin_2.0.9.
If i'm trying to use ABL and push "Start:" to level corners, printer only level Left corners and don't move to Right part.
If i push buttons Left front corner or Left bottom corner, printer Level them, but no reaction for buttons Right front and Right bottom. Looks like they are not active.
I can't figure aout where is problem, in marlin or in TFT firmware.
If echo M111, as i understand terminal send G30 E1 X194 Y194 and G30 E1 X41 Y194 to that points, but in fact don't move BL to that points...please find photo in attached (sorry, i'm new and don't know how to debug by text).
[G30 E1 X41 Y41](https://prnt.sc/19ngoaq)
[G30 E1 X194 Y41](https://prnt.sc/19ngrhn)
[G30 E1 X194 Y194](https://prnt.sc/19ngu8b)
[G30 E1 X41 Y194](https://prnt.sc/19ngwsm)
So I can only Level 2 corners, [photo](https://prnt.sc/19nharm)
If use ABL with mesh, all mesh levels, but right side not so well, i think because can't level corners.

bubulgum, 2021-07-09:
> hi, all.
> I'm using Mega zero 2 with SKR mini E3 v2 and TFT35 E3.
> Firmware KAD_marlin_2.0.9.
> If i'm trying to use ABL and push "Start:" to level corners, printer only level Left corners and don't [...]

same issue with KAD_marlin_2.0.9.1
and BIGTREE_TFT35_V3.0_E3.27.x.bin with config.ini from 08 Jul 2021

radek8, 2021-07-09:
Send Marlin and tft configuration files

bubulgum, 2021-07-10:
> Send Marlin and tft configuration files

[Marlin.zip](https://github.com/bigtreetech/BIGTREETECH-TFT35-V3.0/files/6794641/Marlin.zip)
[TFT.zip](https://github.com/bigtreetech/BIGTREETECH-TFT35-V3.0/files/6794642/TFT.zip)

radek8, 2021-07-10:
Depending on your configuration, the points at position Y194 are not available for the probe.
The problem is in your bed configuration and probe point position.

radek8, 2021-07-10:
The position of the probe on the X axis is 38 mm to the left of the nozzle.
The size of the bed is 220.
You have MAX_SOFTWARE_ENDSTOPS enabled for the X and Y axes.
Max position equals BED_SIZE.
Therefore, the nozzle cannot move beyond the bed

If the probe should move to position X 194
the nozzle would be at position 194 + 38 = 232
So the nozzle would go out of bed and you're forbidden.
 
In touch mode you have defined
level_edge_distance: 20
220-20 + 38 = 238
So the nozzle would go out of bed. But MAX_SOFTWARE_ENDSTOPS will not allow this

radek8, 2021-07-10:
[image]

[image]

[image]

bubulgum, 2021-07-11:
i have commened MAX_SOFTWARE_ENDSTOPS, but nozzle still don't move.
[image]

bubulgum, 2021-07-11:
> > The position of the probe on the X axis is 38 mm to the left of the nozzle.
> > The size of the bed is 220.
> > You have MAX_SOFTWARE_ENDSTOPS enabled for the X and Y axes.
> > Max position equals BED_SIZE.
> > Therefore, the nozzle cannot move beyond the bed
> > If the probe should move to position X 194
> > the nozzle would be at position 194 + 38 = 232
> > So the nozzle would go out of bed and you're forbidden.
> > In touch mode you have defined
> > level_edge_distance: 20
> > 220-20 + 38 = 238
> > So the nozzle would go out of bed. But MAX_SOFTWARE_ENDSTOPS will not allow this
> 
> i have commened MAX_SOFTWARE_ENDSTOPS, but nozzle still don't move to right side.
> [image]

radek8, 2021-07-11:
MAX_SOFTWARE_ENDSTOPS Don't get dirty. It's printer protection.  Edit the coordinates of leveling points. 
Have you also initialized EEPROM?

radek8, 2021-07-11:
Send a photo of the printer with the leveling probe

bubulgum, 2021-07-12:
For what does use for button "Distance" on the screen?
Min 41 and MAX 100, i have change to 50 and it works.

radek8, 2021-07-12:
It is the distance of the measuring point from the edge of the bed

bubulgum, 2021-07-13:
> It is the distance of the measuring point from the edge of the bed
> 
> level_edge_distance: 50
> 220-50 + 38 = 208
> 208<220 Yes, this will work

i'm confused of the formula. 
Because before was 41.
And actually in my situation NOZZLE_TO_PROB_OFFSET = 40.
So 220-41 + 40 = 219...but thats not works.
I have tried 42, 43...49 and only 50+ works.

bubulgum, 2021-07-13:
and one more thing, after setup all corners i have saved to EEPROM, but after i turn off and turn on printer, settings come back to 0.
For the MESH still ok.

radek8, 2021-07-13:
Yes, if you have a probe offset of 40 on the X axis, then 41 should work.
Can you enter the answer to the commands here?
M503
and
M115
Thank you

bubulgum, 2021-07-15:
Sorry, how to log on TXT file? 
if i enable M928 log.txt there is no such file on SD...

radek8, 2021-07-15:
Connect the printer to the computer and use Pronterface

bubulgum, 2021-07-15:
> Connect the printer to the computer and use Pronterface

Thank you

Connecting...
Printer is now online.
 M503
SENDING:M503
echo:  G21    ; Units in mm (mm)
echo:  M149 C ; Units in Celsius
echo:; Steps per unit:
echo: M92 X80.00 Y80.00 Z400.00 E385.00
echo:; Maximum feedrates (units/s):
echo:  M203 X500.00 Y500.00 Z5.00 E25.00
echo:; Maximum Acceleration (units/s2):
echo:  M201 X500.00 Y500.00 Z100.00 E5000.00
echo:; Acceleration (units/s2): P<print_accel> R<retract_accel> T<travel_accel>
echo:  M204 P500.00 R500.00 T500.00
echo:; Advanced: B<min_segment_time_us> S<min_feedrate> T<min_travel_feedrate> J<junc_dev>
echo:  M205 B20000.00 S0.00 T0.00 J0.08
echo:; Home offset:
echo:  M206 X0.00 Y0.00 Z0.00
echo:; Auto Bed Leveling:
echo:  M420 S1 Z10.00
echo:  G29 W I0 J0 Z2.18250
echo:  G29 W I1 J0 Z1.43500
echo:  G29 W I2 J0 Z0.82500
echo:  G29 W I3 J0 Z0.18750
echo:  G29 W I4 J0 Z-0.43000
echo:  G29 W I5 J0 Z-1.00000
echo:  G29 W I6 J0 Z-1.69500
echo:  G29 W I0 J1 Z2.16000
echo:  G29 W I1 J1 Z1.47500
echo:  G29 W I2 J1 Z0.80250
echo:  G29 W I3 J1 Z0.14750
echo:  G29 W I4 J1 Z-0.46000
echo:  G29 W I5 J1 Z-1.11000
echo:  G29 W I6 J1 Z-1.71750
echo:  G29 W I0 J2 Z2.26500
echo:  G29 W I1 J2 Z1.63000
echo:  G29 W I2 J2 Z0.94500
echo:  G29 W I3 J2 Z0.29000
echo:  G29 W I4 J2 Z-0.32500
echo:  G29 W I5 J2 Z-0.98500
echo:  G29 W I6 J2 Z-1.58250
echo:  G29 W I0 J3 Z2.36500
echo:  G29 W I1 J3 Z1.66250
echo:  G29 W I2 J3 Z0.96750
echo:  G29 W I3 J3 Z0.29750
echo:  G29 W I4 J3 Z-0.32500
echo:  G29 W I5 J3 Z-0.79500
echo:  G29 W I6 J3 Z-1.57500
echo:  G29 W I0 J4 Z2.39500
echo:  G29 W I1 J4 Z1.75500
echo:  G29 W I2 J4 Z1.08250
echo:  G29 W I3 J4 Z0.40000
echo:  G29 W I4 J4 Z-0.20500
echo:  G29 W I5 J4 Z-0.78750
echo:  G29 W I6 J4 Z-1.48000
echo:  G29 W I0 J5 Z2.53500
echo:  G29 W I1 J5 Z1.85500
echo:  G29 W I2 J5 Z1.19500
echo:  G29 W I3 J5 Z0.48750
echo:  G29 W I4 J5 Z-0.09750
echo:  G29 W I5 J5 Z-0.72750
echo:  G29 W I6 J5 Z-1.33250
echo:  G29 W I0 J6 Z2.58750
echo:  G29 W I1 J6 Z1.92750
echo:  G29 W I2 J6 Z1.42750
echo:  G29 W I3 J6 Z0.14500
echo:  G29 W I4 J6 Z-0.05250
echo:  G29 W I5 J6 Z-0.66500
echo:  G29 W I6 J6 Z-1.28500
echo:; Material heatup parameters:
echo:  M145 S0 H190.00 B60.00 F0
echo:  M145 S1 H230.00 B70.00 F0
echo:; PID settings:
echo:  M301 P17.72 I1.18 D66.76
echo:  M304 P350.00 I63.00 D452.00
; Controller Fan
echo:  M710 S255 I0 A1 D60 ; (100% 0%)
echo:; Power-Loss Recovery:
echo:  M413 S0
echo:; Z-Probe Offset (mm):
echo:  M851 X-40.00 Y-12.00 Z-0.51
echo:; Stepper driver current:
echo:  M906 X800 Y800 Z800
echo:  M906 T0 E800
echo:; Driver stepping mode:
echo:  M569 S1 X Y Z
echo:  M569 S1 T0 E
echo:; Linear Advance:
echo:  M900 K0.00
echo:; Filament load/unload lengths:
echo:  M603 L400.00 U450.00
echo:; Filament runout sensor:
echo:  M412 S0 D7.00

M115
SENDING:M115
FIRMWARE_NAME:Marlin 2.0.9.1-kad1 (Jul 16 2021 01:02:06) SOURCE_CODE_URL:github.com/kad/Marlin PROTOCOL_VERSION:1.0 MACHINE_TYPE:Mega Zero 2.0 EXTRUDER_COUNT:1 UUID:dbfccd55-ebba-4615-abd7-196aa438f304
Cap:SERIAL_XON_XOFF:0
Cap:BINARY_FILE_TRANSFER:1
Cap:EEPROM:1
Cap:VOLUMETRIC:0
Cap:AUTOREPORT_POS:1
Cap:AUTOREPORT_TEMP:1
Cap:PROGRESS:0
Cap:PRINT_JOB:1
Cap:AUTOLEVEL:1
Cap:RUNOUT:1
Cap:Z_PROBE:1
Cap:LEVELING_DATA:1
Cap:BUILD_PERCENT:1
Cap:SOFTWARE_POWER:1
Cap:TOGGLE_LIGHTS:0
Cap:CASE_LIGHT_BRIGHTNESS:0
Cap:EMERGENCY_PARSER:1
Cap:HOST_ACTION_COMMANDS:1
Cap:PROMPT_SUPPORT:0
Cap:SDCARD:1
Cap:REPEAT:0
Cap:SD_WRITE:1
Cap:AUTOREPORT_SD_STATUS:0
Cap:LONG_FILENAME:1
Cap:THERMAL_PROTECTION:1
Cap:MOTION_MODES:0
Cap:ARCS:1
Cap:BABYSTEPPING:1
Cap:CHAMBER_TEMPERATURE:0
Cap:COOLER_TEMPERATURE:0
Cap:MEATPACK:0

radek8, 2021-07-15:
You can uncomment in configuration_adv.h
  #define M115_GEOMETRY_REPORT
and copy the M115 command again?

bubulgum, 2021-07-15:
> You can uncomment in configuration_adv.h
> #define M115_GEOMETRY_REPORT
> and copy the M115 command again?

Connecting...
Printer is now online.
M115
SENDING:M115
FIRMWARE_NAME:Marlin 2.0.9.1-kad1 (Jul 16 2021 03:23:02) SOURCE_CODE_URL:github.com/kad/Marlin PROTOCOL_VERSION:1.0 MACHINE_TYPE:Mega Zero 2.0 EXTRUDER_COUNT:1 UUID:dbfccd55-ebba-4615-abd7-196aa438f304
Cap:SERIAL_XON_XOFF:0
Cap:BINARY_FILE_TRANSFER:1
Cap:EEPROM:1
Cap:VOLUMETRIC:0
Cap:AUTOREPORT_POS:1
Cap:AUTOREPORT_TEMP:1
Cap:PROGRESS:0
Cap:PRINT_JOB:1
Cap:AUTOLEVEL:1
Cap:RUNOUT:1
Cap:Z_PROBE:1
Cap:LEVELING_DATA:1
Cap:BUILD_PERCENT:1
Cap:SOFTWARE_POWER:1
Cap:TOGGLE_LIGHTS:0
Cap:CASE_LIGHT_BRIGHTNESS:0
Cap:EMERGENCY_PARSER:1
Cap:HOST_ACTION_COMMANDS:1
Cap:PROMPT_SUPPORT:0
Cap:SDCARD:1
Cap:REPEAT:0
Cap:SD_WRITE:1
Cap:AUTOREPORT_SD_STATUS:0
Cap:LONG_FILENAME:1
Cap:THERMAL_PROTECTION:1
Cap:MOTION_MODES:0
Cap:ARCS:1
Cap:BABYSTEPPING:1
Cap:CHAMBER_TEMPERATURE:0
Cap:COOLER_TEMPERATURE:0
Cap:MEATPACK:0
area:{full:{min:{x:-7.00,y:-2.00,z:0.00},max:{x:225.00,y:225.00,z:250.00}},work:{min:{x:-7.00,y:-2.00,z:0.00},max:{x:225.00,y:225.00,z:250.00}}}

> Another question, do you have a calibrated number of steps per mm?

I don't know what does it mean and how to do.
I have just upload firmware, because it exactly for my model with my board...
Printer works well...but for now i found only this kind of problems.

radek8, 2021-07-15:
The configuration looks fine.
home X-axis offset is 0
The probe offset is -40
The work surface is defined on the X axis -7 to 225.
The probe probe should operate at an X-axis coordinate of 184 mm or less.

bubulgum, 2021-07-16:
> Step calibration is essential. You can find the procedure on youtube.

thx, i will

> Does the physical location of the probe match the parameters in the firmware?

yes, this fine and printing fine. 

but i need to setup "Distance" to 50, and every corner i need to up manual by Z, otherwise if move to another corner can scratch bed.

radek8, 2021-07-16:
Tasdy, you're wrong.
In Marlin you have a defined bed as 225x225,
but you have 235x235 defuncted on the touch screen.
You have to define the same (correct) values everywhere.
fix coubor config.ini on 225.
You also have the wrong size_min value.

[image]

[image]

radek8, 2021-07-16:
Also bad value
set 41

[image]

radek8, 2021-07-16:
Pokud máte při přesunu mezi sondováním trysku příliš nízkou a hrozí poškrábání, upravte tyto hodnoty.

[image]

bubulgum, 2021-07-16:
> Tasdy, you're wrong.
> In Marlin you have a defined bed as 225x225,
> but you have 235x235 defuncted on the touch screen.
> You have to define the same (correct) values everywhere.
> fix coubor config.ini on 225.
> You also have the wrong size_min value.
> 
> [image]
> 
> [image]

You are awesome, i missed that part in TFT config.
And i was wrogn about my bed size, you were right 220x220x250.
Now it works perfect!
Thanks a lot!

bubulgum, 2021-07-16:
> Pokud máte při přesunu mezi sondováním trysku příliš nízkou a hrozí poškrábání, upravte tyto hodnoty.
> 
> [image]
> 
> [image]

i have tried to set 10. But this not home.
If i start to L Corner, probe go to each corner without any movement by Z and PROBE only 1 time.

bubulgum, 2021-07-16:
[image]
i have enabled LEVEL_CORNERS_USE_PROB...but stil nothing.
have tried LEVEL_CORNERS_Z_HOP 10...nothing
[image]
[image]

bubulgum, 2021-07-16:
This issue looks like same that i have. But I have tried both and have not fixed this.

radek8, 2021-07-16:
Somehow I'm getting lost in your problem.
You started leveling in Marlin mode,
then you wanted to deal with corner alignment in touch mode.
You are now setting LEVEL_BED_CORNERS. This is another feature in Marlin mode.

radek8, 2021-07-16:
Can you send a video what your problem is?

radek8, 2021-07-16:
If you use the LevelConer function on the touch mode, when you measure the height of the corners of the bed, then after the measurement, the probe should be automatically raised by 10 mm. Unfortunately, an error occurred in the last build.

Try a brick repair

[BIGTREE_TFT35_V3.0_E3.27.x.zip](https://github.com/bigtreetech/BIGTREETECH-TFT35-V3.0/files/6833618/BIGTREE_TFT35_V3.0_E3.27.x.zip)

bubulgum, 2021-07-17:
> If you use the LevelConer function on the touch mode, when you measure the height of the corners of the bed, then after the measurement, the probe should be automatically raised by 10 mm. Unfortunately, an error occurred in the last build.
> 
> Try the fix
> 
> [BIGTREE_TFT35_V3.0_E3.27.x.zip](https://github.com/bigtreetech/BIGTREETECH-TFT35-V3.0/files/6833618/BIGTREE_TFT35_V3.0_E3.27.x.zip)

Thank you, now works perfectly, except 1 thing.
If turn power off and then power on printer, LevelCorners does not save in EEPROM. Need to Level corners again.
Even after leveled corners manually save to EEPROM.

radek8, 2021-07-17:
You balance the corners with screws under the bed. This is the function you're using to see the corner. Storing this value in eeprom is meaningless.

closed 2021-07-19 by bigtreetech