bigtreetech/BIGTREETECH-TFT35-V3.0 #133: WiFi ESP8266-01S can't connect with SKR MINI E3 v2 + TFT35 E3 V3

bubulgum, 2021-07-15:
I have installed software to wifi,and if connect to TFT i can see web interface in browers (have uploaded index.html.gz from here https://github.com/bigtreetech/BIGTREETECH-SKR-PRO-V1.1/tree/master/firmware/esp-01s) and last version of ESP3D
[image]
But still can't see printer...even on display M117 don't show connecting message.
Looks like wifi module inactive with mainboard.
TFT with SKR connected to EXP3-CR10 and TFT cables.
In Marlin i have enabled 
```
#define SERIAL_PORT -1
#define SERIAL_PORT_2 2

//also have tried

#define SERIAL_PORT 3
#define SERIAL_PORT_2 -1

and etc...
```
[Marlin.zip](https://github.com/bigtreetech/BIGTREETECH-TFT35-V3.0/files/6825275/Marlin.zip)
[TFT Config.zip](https://github.com/bigtreetech/BIGTREETECH-TFT35-V3.0/files/6825284/TFT.Config.zip)

radek8, 2021-07-15:
If you connect the WiFi module to the TFT, you have a configuration error.
The serial port for WiFi is disabled.

[image]

bubulgum, 2021-07-15:
> If you connect the WiFi module to the TFT, you have a configuration error.
> The serial port for WiFi is disabled.
Thank you for you replies.

I have enabled:
multi_serial: 1
And Marlin have:
#define SERIAL_PORT -1
#define SERIAL_PORT_2 2

But still no emotion.

radek8, 2021-07-15:
Do you have your Wifi access point correctly defined on the wifi module?

bubulgum, 2021-07-16:
yes, i can connet to the web interface, set up correct Baud Rate.
Looks like Wifi don't have connection with printer, only as power, but not with firmware.

radek8, 2021-07-16:
Your speed is 115200
do you have set

radek8, 2021-07-16:
set in config.ini
serial_always_on:1

bubulgum, 2021-07-16:
[image]

> Your speed is 115200
> do you have set

i have tried 115200, then 250000..nothing.

> set in config.ini
> serial_always_on:1

have set, i think problem with serials, but i don't know where to check exactly right connections.

radek8, 2021-07-16:
> 
> i have tried 115200, then 250000..nothing.
> 

You must not rehearse, you must know what you are setting.

There are a lot of parameters, if you try parameters that are clear, it will take you a long time :-(

bubulgum, 2021-07-31:
can not find where is no connection. spped rate 250000, have enabled ports. but same issue. is it possible to test wifi connection by firmware? because i can connect to wifi through IP, so it is mean wifi connected right and working. but there is no connection with firmware.
Can it be problem because my firmware is Marlin but modify by KAD? is it important?
[configs.zip](https://github.com/bigtreetech/BIGTREETECH-TFT35-V3.0/files/6911113/configs.zip)

radek8, 2021-07-31:
if you have WiFi connected to TFT and TFT communicates with SKR, WIFi should also communicate.
Must have WiFi to TFT serial port enabled (you have it)
The same communication speed must be selected.
If it still does not communicate there, you have Wifi badly connected, or some pin on the serial WiFi connector is damaged.

radek8, 2021-07-31:
I have no experience with the KSD configuration, I cannot judge. But I'm quite confusing for me.

Send a photo connection

bubulgum, 2021-07-31:
I found what was problem...i have uploaded old index.html.gz
now everything wokrs well with this one [https://github.com/luc-github/ESP3D-WEBUI](url)
Thank you for help
[image]

radek8, 2021-08-01:
The web interface of the wifi adapter doesn't suit me very well.
I connected the printer's WIFI adapter as a virtual serial port and I connect to the printer via the Pronterface wifi program.
You can try it, it's functional :-)

KingBain, 2022-12-28:
Just went through this. with SKR Mini E3 v3 and ESP8266-01(clone modules); **not the S variant..** 

I wired everything to the TFT port on the main board 
[image]

I know the chip doesn't like 5v, but that's what I normally feed it and they work fine. 

In my marlin config. 

I haven't changed much, but this is what I'm using 

#define SERIAL_PORT 2
#define BAUDRATE 115200
#define SERIAL_PORT_2 -1