bigtreetech/BIGTREETECH-TFT35-V3.0 #134: Unable to use filament  Runout sensor for dual extruder prints

TechnicalWorld2016, 2021-07-29:
Hi !
I have a 3D printer with dual extruder and a 2 nozzle hot-end.
I am using BigTree Tech SKR PRO V1.2 mother Board and BTT TFT 3.5.
I am printing with 2 nozzles perfectly fine.
I thought of connecting a normal Limit-switch based Filament runout sensor and elabled filament runout in TFT Firmware.

I am able to use it when I am printing with single colour prints..
But when ever I am trying give prints with dual colour, getting an issue as filament runout.

How can I use 2 filament runout sensors with the BTT TFT 35.

I have directly connected 2 filament runout sensors to the mother board and results are perfect with keeping TFT Screen in marlin mode. But I need to use Touch mode.

I need a solution with connecting 2 filament runout sensors to the mother board and useing TFT in touch mode.
Or connecting sensors to the TFT and using them.

So, I request someone to help me with the problem.

Thank you

oldman4U, 2021-07-29:
I would ask the question in the firmware section or do you believe it is a hardware issue?

TechnicalWorld2016, 2021-07-30:
I don't think that there is a problem with marlin firmware because I can connect 2 filament runout sensors  to board and can use the TFT screen in marlin mode successfully..

but when I am connecting a single sensor to the TFT screen and using TFT in Touch mode I am unable to give dual prints because its giving error as filament runout even there is filament present.

At the same time ,It is working fine with single color  prints.

so I need a solution with using 2 filament runout sensors connected with board and using tft in touch mode.
Or directly connectiong Filament runout sensors to TFT.

oldman4U, 2021-07-30:
Who talked about Marlin?

TechnicalWorld2016, 2021-07-30:
OK 
In TFT firmware there is only option for single filament runout sensor.
I have enabled it and getting results with single prints. But there is error with dual prints..
I am unable to find where is the issue..

oldman4U, 2021-07-30:
Thats why I recommend to close this ticket here and open a new one in the
related repository.

TechnicalWorld2016, 2021-07-30:
Actually my problem is related to Bigtree tech TFT 35 v3.0
So I have created a ticket here.

oldman4U, 2021-07-30:
OK.