bigtreetech/BIGTREETECH-TFT35-V3.0 #137: SKR 2 with TFT 35 V3 - No Printer Attached.

JaseGill, 2021-08-06:
I have an issue with my TFT35 V3.0 connected to an SKR V2.
It works fine in Marlin Mode and I can home, bed level and BLTouch even works so I believe my config is good.

I have tried .027 latest Firmware from the GITHUB as well as compiling it myself with both 115200 and 250000 Baud Rates along withe Marlin 2.0.9.1 with Baud Rates set to same to match.

In the TFT Configuration. h I have:
#define BAUDRATE 4  // Default: 5

In the TFT Config.ini I have:
baudrate:4

My Serial Ports in Marlin for the SKR 2 are set to:
#define SERIAL_PORT -1
#define BAUDRATE 115200

#define SERIAL_PORT_2 2
#define BAUDRATE_2 115200 

//#define SERIAL_PORT_3 1
//#define BAUDRATE_3 115200 

I have tried 115200 and 250000 for both the TFT and the SKR 2, with the same response 'No Printer Attached'.  I have also using the Touchscreen menu's changed the BaudRate to match the config I am using.  Again no change.

Does anyone know anything else I can try?
Thanks.

radek8, 2021-08-06:
Your configuration is bad

#define BAUDRATE 4 // Výchozí: 5
This sets the speed of the anTFT 57600
You have set 115200 on the SKR
You must set the same speed

JaseGill, 2021-08-07:
Just updated BAUDRATE to 5.
No Change.  Same response, 'No Printer Attached.

JaseGill, 2021-08-07:
I found the issue.
Was a cable break in the TFT Cable.  Once repaired it is now funtioning.

closed 2021-08-07 by JaseGill