bigtreetech/BIGTREETECH-TFT35-V3.0 #138: Cannot see mainboard TF card since upgraded to the latest version on TFT 35.

PMesterur, 2021-09-22:
I have SKR Mini e3 v2 board. I received the TFT 35 with firmware B1.26.
I upgraded to B1.27 a few days ago.
It works a lot better, but since then, when I press Print on the screen, I can see only print from USB and print from SD function, the mainboard TF disappeared. 
When I switch to Marlin mode, I can open the TF connected to the mainboard and I can see the content.
Does anybody has any idea, how can I fix that?

radek8, 2021-09-22:
Please inserted marlin and TFT configuration files

PMesterur, 2021-09-22:
[BIGTREE_TFT35_V3.0.27.x.zip](https://github.com/bigtreetech/BIGTREETECH-TFT35-V3.0/files/7213674/BIGTREE_TFT35_V3.0.27.x.zip)

I have no marlin configuration file, I use what I received with the board: Marlin bugfix-2.0.x (GitHub)

radek8, 2021-09-22:
It's hard to find fault if you don't
config.ini for TFT
and configuration.h, configuration_adv.h for Marlin

radek8, 2021-09-22:
Did you at least use the config.ini file for TFT configuration?

PMesterur, 2021-09-22:
Yes, sorry, I forgot that
[config.zip](https://github.com/bigtreetech/BIGTREETECH-TFT35-V3.0/files/7213858/config.zip)
.

radek8, 2021-09-22:
In config, ini I would change a few parameters, but when it comes to reading the SD card on the board, it's fine.
In Marlin, you must ensure that the following parameters are enabled.
  SDSUPPORT (in Configuration.h)
  LONG_FILENAME_HOST_SUPPORT (in Configuration_adv.h)
  AUTO_REPORT_SD_STATUS (in Configuration_adv.h)
  SDCARD_CONNECTION ONBOARD (in Configuration_adv.h)

radek8, 2021-09-22:
Set in config.ini
persistent_info: 1
layer_disp_type: 2
notification_m117: 1

It will not solve your problem, but you will see more data on the display. :-)

PMesterur, 2021-09-25:
I managed to solve it with your help, thank you.

closed 2021-09-25 by PMesterur