bigtreetech/BIGTREETECH-TFT35-V3.0 #140: "No printer attached!" Error with btt SKR 2

Matze1811, 2021-11-13:
Hi - I've got a problem trying to connect a TFT 35 to a skr 2 that I can't fix.

My setup:

BTT SKR 2
Marlin 2.0.9.1
BTT TFT35 V3.0
Symptoms:

Marlin mode works
Touchscreen mode does not work 'No printer attached!' - No hotend data, no commands work, etc.

Things I've tried:

Ensuring SERIAL_PORT(s) and baud rate are set in Marlin. I've tried 0/-1, -1/0. Baud is 115200
Ensuring serial on TFT is 115200
Ensuring SDSUPPORT is defined
Trying 250000 and 57600 baud rate (matching Marlin and TFT each time)
checking cable continuity from TFT board pins to SKR pins (no issues)
Checking continuity on TFT pins between themselves (there is another issue that said there is a dodgy batch of TFT with internal shorts)
switching Rx/Tx wires to see if there was a mismatch

ANY Ideas?
THX
Matze

radek8, 2021-11-13:
Please insert Marlin configuration file (configuration.h and configuration_adv.h) and TFT configuration (config.ini)
send photos of cable connections between SKR and TFT.
Thank you

radek8, 2021-11-13:
> Věci, které jsem vyzkoušel:
> 
> Zajištění SERIAL_PORT(ů) a přenosová rychlost jsou nastaveny v Marlin. Zkoušel jsem 0/-1, -1/0. Baud je 115200 
> Nějaké nápady? THX Matze

CAUTION! 
SKR2 uses TFT port 1 to communicate
so you have to set it
#define SERIAL_PORT 1
The speed must be the same on both devices. Try 115200

Matze1811, 2021-11-13:
Hi there,
I've already tried everything. The error persists. attached the configuration.h

Matze1811, 2021-11-13:
[Configuration.odf](https://github.com/bigtreetech/BIGTREETECH-TFT35-V3.0/files/7532222/Configuration.odf)

Matze1811, 2021-11-13:
[Configuration_adv.odf.txt](https://github.com/bigtreetech/BIGTREETECH-TFT35-V3.0/files/7532237/Configuration_adv.odf.txt)
[Photos (6).zip](https://github.com/bigtreetech/BIGTREETECH-TFT35-V3.0/files/7532242/Photos.6.zip)

radek8, 2021-11-13:
uncomment

#define M114_DETAIL   
#define REPORT_FAN_CHANGE
#define M115_GEOMETRY_REPORT
#define AUTO_REPORT_POSITION
#define HOST_ACTION_COMMANDS
#define HOST_PAUSE_M76
#define HOST_PROMPT_SUPPORT
#define HYBRID_THRESHOLD
#define MONITOR_DRIVER_STATUS
#define SERIAL_FLOAT_PRECISION 4
#define EMERGENCY_PARSER
#define AUTO_REPORT_SD_STATUS
#define LONG_FILENAME_HOST_SUPPORT
#define UTF_FILENAME_SUPPORT

radek8, 2021-11-13:
you did not send a TFT configuration
send config.ini

radek8, 2021-11-13:
Update FW TFT to the latest versions

Matze1811, 2021-11-14:
HI,
[config.odf.txt](https://github.com/bigtreetech/BIGTREETECH-TFT35-V3.0/files/7533212/config.odf.txt)

here the config.ini

I had already done an update of the firmware to the last version. TFT 35 and SKR Board are new!

radek8, 2021-11-14:
Konfigurace TFT je v pořádku.
Kabelové spoje také vypadají dobře.
Upravte konfiguraci portu v Marlin a odkomentujte parametry, které jsem napsal.
Vyzkoušejte, zda připojení displeje funguje v dotykovém režimu.

Matze1811, 2021-11-15:
[Configuration_adv.odf.txt](https://github.com/bigtreetech/BIGTREETECH-TFT35-V3.0/files/7535749/Configuration_adv.odf.txt)

Hello, all parameters that you wrote are already taken out in the configuration.adv .h.

The problem is the same: "Cannot connect to the printer"

radek8, 2021-11-15:
For the parameters I wrote you have to uncomment (delete characters //)
They are still commented on in the file you sent.

radek8, 2021-11-15:
Do you have serial port 1 set up in Marlin?

radek8, 2021-11-15:
Looks like your screen is receiving messages from Marlin?
This is a message that the engine controllers are not communicating (no TMC drivers are fitted). 

[image]

Matze1811, 2021-11-15:
[Configuration_adv.odf.txt](https://github.com/bigtreetech/BIGTREETECH-TFT35-V3.0/files/7539586/Configuration_adv.odf.txt)
[Configuration.odf.txt](https://github.com/bigtreetech/BIGTREETECH-TFT35-V3.0/files/7539587/Configuration.odf.txt)

Hello, I took the // out. Enclosed the configs. The error pattern is the same.

I now have 2 pieces of SKR2 boards. The photo I took is the board on my desk. No TM2209s are fitted here. Both boards, including the one equipped with TMC 2209, show this error.

The setting of serial 1 should be in order, right?

radek8, 2021-11-15:
The configuration is now OK.
The speed on both devices is set to 115200
You wrote that you measured the serial cable and it's fine too.
Do you have 2 SKR 2 and the display will not connect to both?
If so, one of the Rx or Tx pins on the display's serial port is probably damaged.
You are not the only one who has such a problem.
So we can try to redirect the communication to the UART3 port of the display.
It is one of two solutions. The second option is to complain to TFT sellers

radek8, 2021-11-15:
look here
https://github.com/bigtreetech/BIGTREETECH-TFT35-V3.0/issues/120

radek8, 2021-11-16:
If you start the terminal on the display.
You send a G28 command.
You turn on all incoming communication (the ON icon next to the back button will light up).
Is the G28 command executed?
Is there any incoming traffic?

radek8, 2021-11-16:
With that SKR2 record, you are unlucky to have bought it.
This model didn't do well :-(
It brings too many problems.

Matze1811, 2021-11-17:
**Hi there,
the command G28 returns an M503 S0.
There must be some kind of connection because the message "all low" does not appear on the printer where the TMC2209 are plugged.
I need new hardware to switch to another port for TX RX. But I don't think it's that, we can give it a try. If so, can you send me new firmware?

radek8, 2021-11-17:
Load this FW into the display

[UART3  FW.zip](https://github.com/bigtreetech/BIGTREETECH-TFT35-V3.0/files/7556895/UART3.FW.zip)

If you have the original FW too old, you will also need to update the icons and font.
Disconnect the RX and TX contacts from the connector on the display. Connect these connectors to the RX and TX connectors of the UART3 connector.
If the display does not connect, try swapping the RX and TX contacts.

radek8, 2021-11-20:
some news?

Matze1811, 2021-11-21:
Hi there,

I copied the corresponding files to the SD card. but this error message comes u
[image]
p!

radek8, 2021-11-21:
That's ok.  Also update the icons.

radek8, 2021-11-21:
Attention. 3 days ago de some icons renamed. Use icons 4 days old

radek8, 2021-11-21:
Choose a theme and copy the TFTxx folder to your SD card.
restart the display

https://github.com/bigtreetech/BIGTREETECH-TouchScreenFirmware/tree/78c41d021ad0cb1330eabaa8c90e6c95b6824dd4/Copy%20to%20SD%20Card%20root%20directory%20to%20update/THEME_Unified%20Menu%20Material%20theme

nettlecat, 2021-11-21:
im having the same issue but i think its on the SK2 itself. i have all firmware and configs set from this thread but even using the UART3 im not getting any results. could my tx pins on the sk2 be shot? is there a way around that?

nettlecat, 2021-11-21:
resolved my issue. may have been that i had the ribbon cables in. may have been poorly seated wires. but it now works with UART3. thx!

mAd-DaWg, 2021-11-28:
I'm having a similar issue. I can send and receive commands on the tft's terminal though. M115 responds properly. Marlin mode works fine. tried disconnecting the ribbon cables and use tft mode, and M115 still sent and received fine, but the screen still says "No printer attached"

Also tried reflashing the tft's firmware, and using a reset.txt, but still no change

radek8, 2021-11-28:
@mAd-DaWg
Send the marlin and TFT configuration files. 
Send photos of wired cables EXP and TFT

radek8, 2021-11-28:
From his date you have FW Marlin anf TFT

radek8, 2021-11-28:
> From our date you have FW Marlin anf TFT?

mAd-DaWg, 2021-11-28:
Marlin is 2.0.9.2 (from the BTT skr2 repo), with some changes for cnc routing. TFT is v27(flashed within the last 3 days)
[firmware configurations.zip](https://github.com/bigtreetech/BIGTREETECH-TFT35-V3.0/files/7613663/firmware.configurations.zip)

The wiring:
[image]

[image]

Response from M503:
[image]
[image]

[image]
[image]
[image]

radek8, 2021-11-28:
A few days ago there was a bug in the FW display that prevented the connection to the Marlin board https://github.com/bigtreetech/BIGTREETECH-TouchScreenFirmware/issues/2255.
It has already been removed. Try the last FW

mAd-DaWg, 2021-11-28:
@radek8 Flashed the new FW and its working! Thank you so much!

Matze1811, 2021-11-30:
works... THX

closed 2021-11-30 by Matze1811