bigtreetech/BIGTREETECH-TFT35-V3.0 #141: Bed Leveling Position Offset

mpheyse, 2021-12-03:
The best and fastest way I've found to level a bed manually is to go to a auto-parts store and buy a $6 Stainless Steel feeler gauge.

Its quick, Its fast, and accurate. I use the 12 Mil (0.3mm) leaf.

However with the TFT35, the Level Positions go to Z0 every time. So I click the Level-Position, then leave leveling go move, set to 0.1mm and press Z++ three times. (every position, and I always go around at least twice).

So I would like to see a setting in the settings menu for an optional Z Position Offset  (default to zero to maintain current Functionality)

As a Sub request as changeable offset for X and Y (for points 2,3,4) From point 1(Manual Bed Leveling), for smaller or larger beds than the default (200mm?) in the settings menu would also be nice, but it not as needed

radek8, 2021-12-03:
Do you use the current FW in TFT?

mpheyse, 2022-02-17:
Nope and that had it.

closed 2022-02-17 by mpheyse