bigtreetech/BIGTREETECH-TFT35-V3.0 #142: Tft35 v3.0 add mixer settings?

hwystitch, 2021-12-04:
I can control the mixer settings in marlin mode but I dont see anything in touchscreen mode. Is there any way to enable this??

radek8, 2021-12-05:
Not implemented

radek8, 2021-12-18:
Try to submit your request here.
https://github.com/bigtreetech/BIGTREETECH-TouchScreenFirmware/issues
Maybe some programmer will take care of it.

hwystitch, 2021-12-18:
thanks for info, but I am abandoning the BTT main skr 2 and moving up to a fystec board. Far too many issues with the big tree tech board. The thing keeps crashing and you have to wait 2 or more hours with the board off before it will work again and then its hit or miss if the tft will work with it. Also octoprint is now having issues with this board. Cant get the wifi bought from big tree to work with the board either. Big tree tech was of little help trying to get anything working. so going to move on to another board maker, hopefully they will have a better working board.

radek8, 2021-12-23:
So can you close the ticket?
Thank you