bigtreetech/BIGTREETECH-TFT35-V3.0 #143: BTT TFT35 v3 with skr 2 unknown command M566

Phill1337, 2022-01-15:
Hi,

My BTT TFT35 v3 returns unknown command M566 with some x and y values (1800 or 420) behind it very frequently while printing (i think every time it changes direction or something). I assume it has something to do with marlin  but i dont know why my board returns m566 all the time...

The command isnt even on the marlin gcode page (nor in the gcode im  running)...

Regards,
Phillip

radek8, 2022-01-15:
Please send a specific error and send the file you are printing.

Phill1337, 2022-01-15:
[CFFFP_3DBenchy.txt](https://github.com/bigtreetech/BIGTREETECH-TFT35-V3.0/files/7874896/CFFFP_3DBenchy.txt)

I changed to txt as github doesnt want me uploading gcode for some reason...

The specific error is: "unknown command: M566 X1800 Y1800"

Phill1337, 2022-01-15:
The exact error massage is : "unknown command M566 x1800 y1800"

And i just saw that my gcode does actually contain m566 commands... dont know how i missed that before -.- So how do i get my tft to know that command?

closed 2022-01-15 by Phill1337