bigtreetech/BIGTREETECH-TFT35-V3.0 #144: TFT35 screen remains blank until EXP-2 is disconnected

scott-pett, 2022-01-15:
My TFT35 V3's screen will remain blank until the EXP-2 connector is removed and replaced. It will then function correctly.
Also - When updating the firmware from the onboard SD card, this will not function correctly until the EXP-2 connector is removed.
IF the reset button is pressed, the screen goes blank until EXP-2 is removed and replaced.

I have tried several different firmware revs and this happens with both Klipper and Marlin firmware on my BTT Octopus board.

My RepRap Discount Full Graphics display works fine with the same cables and firmwares.

radek8, 2022-01-15:
Is Marlin mode working?
Looks like your connector is rotated 180 °.
Can I send a photo of the connection?
In order to obtain the orientation of the connectors on both sides.

scott-pett, 2022-01-17:
Hi, Thanks Radek8,

For some reason best known to itself, after being switched off overnight, when I powered the board up, the TFT35 fired into life as normal and has not repeated the lockup syndrome. Touch and MArlin mode work OK, and I can access the onboard SD card via Marlin mode, but not via touch mode.

However, I believe my problems stem from my TMC2130 stepper drivers which had to be modified to work on my other controller board - the MKS Rumba32 which has one of the SPI tracks misrouted.

I will do further testing once I've unmodified the TMC2130s. But for the moment, I will close this ticket.

closed 2022-01-17 by scott-pett