bigtreetech/BIGTREETECH-TFT35-V3.0 #145: No printer attached

A1ex7812, 2022-01-16:
[image]

A1ex7812, 2022-01-16:
After a tedious research and opinions from other 3D printer users, I decided to purchased the cr touch for my Creality 3D printer Ender 3. I followed the instructions and downlow the software from the creality website and load it to my SKR mini E3. After this, the error "No printer attached" never disappeared when I powered up my printer, and when I switched to the Marlin Mode, it always stays black. I have update the firmware of my TFT as well for my mother board, without any success. The picture attached to this message is the current version of my touch display.
The led "D6" from the TFT board is a steady red, as for the "Power" and "Status" from the SKR board.

A1ex7812, 2022-01-18:
[image]
[image]
[image]
[image]

[Files.zip](https://github.com/bigtreetech/BIGTREETECH-TFT35-V3.0/files/7885503/Files.zip)

radek8, 2022-01-18:
You have an error in the display configuration in the marlin configuration. That's why Marlin mode doesn't work for you.
In the evening (when I'm sitting at the PC) I will check the files and write the necessary adjustments.
Please send a new photo of the display to see the markings on the black 5-pin cable (RS232)
Thank you

radek8, 2022-01-18:
I'm looking at the configuration you sent.
Are you kidding me?
In the description you write that you have the SKR mini E3, but you do not state the versions.
In the photo is SKR mine E3 V2.0
The configuration you sent is a very old configuration for SKR V1.4 Turbo. This will never work for you.

radek8, 2022-01-18:
I don't know what you have connected to the Z-Probe connector.
If it is BLTouch and the color of the cables corresponds to the standard, then you have the connection upside down and there is a risk of a short circuit of the 5V power supply, if something has not burned yet.

A1ex7812, 2022-01-19:
[image]

I'll make sure to select the correct version. 

I plug the Z-Probe back just to see if something change.

Is not a BL Touch, is a CR Touch.
[image]
[image]

radek8, 2022-01-19:
You really have a bad CRTouch involved.
you must rotate the connector 180 °.
Look at the pinch.

[image]

radek8, 2022-01-19:
Why do you upgrade your printer when you don't know anything about it. Sooner or later you will destroy something. First, study the theory and some instructions. Find the wiring diagrams and before you connect something, make sure you connect it correctly.
Where did you connect the fan on the extruder? Where did you connect the fan that cools the electronics?
Submit the new Marlin configuration you want to use.

radek8, 2022-01-19:
The cables between the display and the SKR are connected correctly.
uf :-)

A1ex7812, 2022-01-19:
I don't know if you are seeing the picture like I see it, but the wiring is
right. First, is a poka-yoke connection, meaning it can only go one way.
Second, according to the circuit wiring from the manual, the white is
ground, black is voltage, yellow is "in", red is ground and blue is "out".
If you turn my picture 180° to match the diagram, the CR touch is
connected like it is supposed to be.
[image: image.png]

radek8, 2022-01-19:
Yes, I'm not at your printer, but I see your photos and I see the difference.
Can you find the difference too? 4 differences? Blue, red, white and black difference.

[image]

radek8, 2022-01-19:
The standard servo connection (LBTouch is a servo) is
GND -   brown or red
5V -      Blue
signal - yellow or orange
black -   GND analog
white -  probe output

If you want to keep the color scheme, swap the wires on the connector that is on the probe.

A1ex7812, 2022-01-19:
The probe is one way in as well, but you know what, thanks. I'll take it
from here, thank you for your time.

closed 2022-01-19 by A1ex7812

radek8, 2022-01-19:
Do not rely on the fact that the connector has a lock and cannot be connected in reverse. The error may be made or the original board from the Creality connector may be reversed.
For example, BTT and MKS boards have EXP connectors rotated 180 ° from each other

radek8, 2022-01-19:
I'm sorry you don't believe me.
I hope you figure it out before you operate Marlin and the activated probe will cause a short circuit.
Happy press.

ptroy1, 2024-03-09:
Hi A1ex7812

I seem to have a similar situation as you.  I have an Ender-3 Neo that comes with a CR touch.  I changed my motherboard to a Big Tree Tech SKR Mini E3 V2.  Marlin works, but when I auto home, it starts properly, but even though the probe retracts when it hits the bed, it doesn't stop the downward motion.

Did you have to change your wiring - my wiring is the same as what you pictured here?  And if not, how did you resolve your issue.  

Thanks . . .

Phil

RabidNative, 2024-03-21:
Hey there ptroy1
I have two enders running Mini E3 v2.  The only way I was able to get the CRTOUCH to work correctly with them with 2.1.x.x marlin has been to split the 5 pin probe connector and use a 3+2 setup.  Both of these printers have worked with the 5pin probe setup for 2+ years using preconfigured firmware from BTT as well as customer configurations that compiled by me with older versions of Marlin.  

I have been unable to locate a working solution to compile a firmware for these boards with CRTouch and new releases of marlin.

ptroy1, 2024-03-26:
Hi

I managed to get things to work (mostly) by switching from Marlin to Klipper.  I have no problem with auto home or auto level, and I did not have to change the wiring as you suggested.

FYI only, BigTreeTech was able to send me a compiled Marlin file that auto homed properly.  But I could not change any of the settings in Marlin because the code had been pre-compiled.  And then when I went to do an automatic leveling, if failed because the probe was off the side of the bed.  I (pretty much) begged them for more help because someone had obviously figured out a solution to the problem, but to no avail.

The klipper solution is working, but now I am having problems with my extruder (it's stuttering), and with setting z-offset.  I may need a new extruder, but I shouldbe able to figure out the z offset.

Thank you veyr much for responding.

RabidNative, 2024-03-26:
Your experience with BTT and requesting firmware is in line with what I have seen elsewhere.  They will provide compiled firmware for the board but will not release the config data.  When attempting to compile my own using newer versions of Marlin,  they do not operate correctly.  

Changing to klipper is not a solution to this issue, it just avoids the entire issue buy using a completely different system.  However, this does seem like the most likely path with BTT no longer supporting these older boards