bigtreetech/BIGTREETECH-TFT35-V3.0 #147: TFT Screen & Printing Failures

Ron234, 2022-04-22:
Hello I have installed a SKR Mini E3 V3.0 it works fine with standard Ender screen on my Ender 5 Pro.  But when I fit an TFT 35 e3 v3 the print stops printing, this happens at various times during printing. Each time I have to turn off printer to remove the print, I have tried new sd cards and Usb memory. But nothing works so have to go back to original screen?

radek8, 2022-04-23:
I don't really understand your problem.
Can you describe it more?

Ron234, 2022-04-24:
Hello what I am saying is that when I try to print something, the printer just stops. I can’t get it to start again. This happens at different times in every print. I have tried new and old cards and Usb memories. I have also updated TFT firmware. When I go back to the original Ender 5 pro screen it prints great. But will not with the TFT screen, even in Marlin mode. I have a SKR Mini V3 E3.0 mainboard. Hopefully this is more helpful.

Ron

Ron234, 2022-04-24:
A pic of what is happening 

Ron

radek8, 2022-04-24:
Are you saying the printer freezes while printing?
Do you see an error message? If so, send a photo.

Click on Menu-Settings-Info and send a screenshot.

Ron234, 2022-04-24:
Yes it freezes up. I had no error messages. But after another try at TFT firmware I get this.

Ron

mayder, 2022-11-15:
[](https://hamy.com.br/video_display.mp4)

This is the same problem.

If I print from the motherboard I don't get any errors.
But using the display this happens intermittently.

See in the video the failures in the printing, each failure was a moment when it stopped.

mayder, 2022-11-15:
[https://hamy.com.br/video_display.mp4](url)
This is the same problem.

If I print from the motherboard I don't get any errors.
But using the display this happens intermittently.
See in the video the failures in the printing, each failure was a moment when it stopped.

fherreragit, 2022-12-06:
Hello. I have an all original BIQU-B1. And I have intermittent problems with prints made from the USB and SD ports on the LCD card. If I use the SD port on the motherboard or if I print from octoprint I have no problems. I've already tested with different cards and pendrives, the problem persists.
The problem is that the 3 axes freeze, I can't move the hotend or cancel printing. The bed and the hotend remain warm.
The LCD doesn't freeze, but it doesn't send any more commands to the motherboard. When the problem occurs there is nothing to do other than restart the printer and consequently lose printing.
I see a lot of people with the same problem and no apparent solution.
In short, prints from the LCD fail, from the motherboard they work fine. Even GCode.
It seems that the LCD loses connection with the motherboard.
My LCD versions BTT B1 TFT35 V3.0 - Firmware: TFT35_B1_V3.0.26.x Aug 18 2020 - Motherboard: BTT SKR V1.4 - Firmware: Marlin 2.0.6 (Aug 24 2020 15:11:48).
Is there anything that can be done to fix the problem?

radek8, 2022-12-06:
Your screen has a very old FW (March 27, 2022).
She would need to update.