bigtreetech/BIGTREETECH-TFT35-V3.0 #148: Black Screen

lickmyballzack, 2022-06-07:
Hey! I apologize in advance I work and may not reply to this thread right away.

I have a BTT TFT35 v3.0 and SKR V1.4 TURBO. I updated marlin to the latest version the way I normally do by throwing it into the USB mode and directly compiling it into my onboard micro. However, since the update it seemed to have bricked my screen where I only receive a black screen where the backlight appears to be on. I completely disassembled the printer to just the board (steppers removed) and the screen that's connected to just the RS232 black 5 pin connector (I am certain that it is connected properly because I can use the reset button still, the red and green LED from when the SD card is inserted are powered, and occasionally I can hold the knob down and hear a clicking sound).

I've attempted to reflash a recompiled fresh TFT firmware from the main firmware branch to no resolve. I attempted to use the reset.txt file method to try to hard reset it but even then the calibration screen never comes up. I read on an old issue here that butchering an old usb cord to only power the screen to insert the txt file may work but it did not for me (yes, I did it correctly, the red LED and green one in the back of the screen powered).

Bricked screen or is there saving me $60?

radek8, 2022-06-07:
I have a few questions:
1. Are you saying the screen is still black?
2. Is it possible to switch between simulated Merlin mode and touch mode?
3. Are at least the icons in touch mode displayed?
4. If you insert a new TFT FW on the SD card and reset. Will the bin file be renamed?
5. Can you insert photos of the display and cable connections here?

lickmyballzack, 2022-06-08:
Through it all, when there’s power to the screen it simply doesn’t display anything. I cannot switch between modes as holding in the button simply just gives a clicking noise. No icons whatsoever are present and when I touch the screen nothing seems to happen. When an SD card is inserted, the green LED on the back turns on, however, it doesn’t seem to actually take within the device because the firmware isn’t updated to remove the .bin.

lickmyballzack, 2022-06-08:
[image]
the “black” screen it may be hard to notice but there is like a backlight on and the screen is lit up. 

[image]
The back of the display, the red LED should signal power to the display and the green signals the SD card. 

[image]
My mobo connection. I undid the RST wire as it has no impact over display functionality and to show the orientation. This is the correct orientation as matched with the back of the board and as how I’ve always added it. I added the EXP1 and 2 ribbons just to show that there’s no functionality in the screen whatsoever.

radek8, 2022-06-08:
Do you have cables EXP1 and 2 connected well?
According to the photo, they are crossed.
EXP1 is connected to EXP2 and vice versa.

radek8, 2022-06-08:
Is the serial cable connected properly?
Take a picture of the cable marking on the SKR side

lickmyballzack, 2022-06-08:
[image]

lickmyballzack, 2022-06-08:
[image]

lickmyballzack, 2022-06-08:
For clarification, the serial cable is connected to the TFT prongs on the mobo. The reset pin is all the way to the left (according to the orientation of my pic)

radek8, 2022-06-08:
OK thank you. The connection seems to be fine.
It looks like a damaged FW in the TFT.
Upload a new FW. If you can't, use another SD card.
Try more SD cards with FAT32 format

lickmyballzack, 2022-06-08:
I’ve attempted to do this. Even went as far as trying the reset.txt to no prevail

radek8, 2022-06-08:
If you connect the display. Can you see that the backlight of the display lights up?

lickmyballzack, 2022-06-09:
> If the display backlight is OK and you have already tried all the options, you can try to load a new bootloader.
> 
> https://github.com/bigtreetech/BIGTREETECH-TouchScreenFirmware/tree/master/Bootloaders/TFT35%20V3.0%20Bootloader%20fix
> 
> #20

Just drop these all into the SD card and boot it?

radek8, 2022-06-09:
No.
You need to connect it to a PC via STLink and use the bootloader upload software.

radek8, 2022-06-09:
you need this HW
[image]

radek8, 2022-06-09:
https://www.youtube.com/watch?v=aTEo-VW_UBY&t=10s

lickmyballzack, 2022-06-10:
> No. You need to connect it to a PC via STLink and use the bootloader upload software.
> 

Yeah I don’t have this STLink. Guessing the USB port on the display would be no bueno either? I got the second half of the USB I butchered laying around and some soldering skills. Could I potentially just make one?