bigtreetech/BIGTREETECH-TFT35-V3.0 #149: Tft35 resets printer when i turn the rotary encoder.

waynemurray74, 2022-06-16:
When I try to turn the rotary encoder, the printer and tft reset. Happens in marlin mode or touchscreen mode. I can use the touchscreen, because I dont need to turn it, but marlin mode is basically non functional.

radek8, 2022-06-16:
Can you give more details?
Display type, printer control board type, used FW.
Attaching configuration files.
Connection of connecting cables (preferably a photo).

waynemurray74, 2022-06-16:
It is the TFT35 V3 , paired with an SKR2  REV.B STM32f429. Using both, RS232 and EXP1+EXP2, connections. they are connected correctly. symptom happens, even if I disconnect the RESET PIN. 
[config.txt](https://github.com/bigtreetech/BIGTREETECH-TFT35-V3.0/files/8923353/config.txt)

radek8, 2022-06-17:
Does the restart occur when I turn left, right, or click?
connector EXP2 and EXP3 also contain a reset pin.
Are you really properly involved?
Send a photo of the connection.

waynemurray74, 2022-06-17:
so, you telling me that there was a reset on exp2 as well helped, since I don't use exp3. upon inspection, i realized that it was plugged in backwards. i rotated it, and everything is working ass it should now. issue resolved! thanks so much for your help!!!

closed 2022-06-17 by waynemurray74