bigtreetech/BIGTREETECH-TFT35-V3.0 #150: TFT35 V3 E3 skr mini V3 E3 Zits blobs

Kam84db, 2022-07-14:
Hi everyone 👋 I have to ask, maby someone will help me with that issue. I have checked every topic about it, I have tried every possible configuration, different ports, different budrates, still the same. Baudrade is the same on the tft and mainboard. 
GD Tft35 v3 e3  GD_TFT35_V3.0_E327 firmware
Skr mini v3 e3
Marlin 2.1
Baudrade 115200
Port 0" 2
Port 2" - 1

I've tried using higher baud 250000 but it freeze the print, fans are working, screen will respond but not the mainboard. Occasionally it dosent freeze but printer is doing wird things - for example during printing for no rizon going in a straight line to the top of x axis and back, or y axis and back, and after that prints again. 
When the baud rade is 115200 it prints ok but it looks like micro stops,  sometimes it sops for a second causing zits/blobs on the print. It only happens when printing from the screen USB/SD. Printing from the skr mini main board works fine. I've tried swapping ports, different marlin versions and no effect. Printing from tfts usb or SD is causing zits on rounded surfaces all the time. Any ideas?

radek8, 2022-07-14:
What print speed did you set when slicing?

radek8, 2022-07-14:
You write that the problem is the same on SD and USB.
In the USB connector, are you using a USB to SD adapter with the same SD card, or a separate USB Disk?
There could be a problem with the SD card.

Kam84db, 2022-07-14:
Hi, speed is 30 outer 50 inner wall top bottom 50, cura salicer. I have the exact firmware that you showed. I've tried micro sd that worked on the main board slot with the adapter to the screens sd slot, with the adapter to usb, normal usb drive and the effect is always the same. Today i wrapped the tft cable with tin foil, and moved the screen farther from the printer, but the result is always the same - zits. Maby the screen is faulty, or the main board?

Kam84db, 2022-07-14:
Those zits looks exactly like here:
https://github.com/bigtreetech/BIGTREETECH-TouchScreenFirmware/issues/282

Kam84db, 2022-08-11:
Ok so I've tried tft24, same problem. I guess something is wrong with the motherboard..??? God only knows why i can't use usb on the screen.