bigtreetech/BIGTREETECH-TFT35-V3.0 #153: TFT35 V3.0 E3 display freezes

Jannomag, 2022-08-03:
Hi,
today I wanted to start a print as usual but I noticed that the display was frozen. The beep sound for touch actions was still there and some areas of the screen changed when I tapped on something, but not everything.

So I tried to flash the latest version, which I did a few weeks ago (and since then everything worked fine, until now). I put everythin on the SD card and the updater started.
While updating icons the screen freezed again at icon 51.
I tried to redo everythin many times, I tried older firmware versions but nothing worked.

Also, when I press the reset button, the screen turns white and nothing happens.

Marlin mode doesn't work either.

When I boot up the printer, the display seems to be working fine for a few seconds until it freezes again.

I didn't change anything before this happened. I was printing nearly every day without any issues for the past 3 weeks,

Any ideas what it could be?

radek8, 2022-08-03:
If you try the update again, does the icon update always freeze in the same place?

Jannomag, 2022-08-03:
> If you try the update again, does the icon update always freeze in the same place?

The last few times it did but there were also some differences before. 
I also tried 3 different SD cards, btw

radek8, 2022-08-03:
It looks like a defective eeprom memory in which the font and icons are stored.
Replacing it shouldn't be too complicated.
I shared a video of her replacement on Youtube.
This is the same type of memory in which the BIOS is stored on some older PC boards.
If you have an old PC, you could recycle the chip :-)

https://www.youtube.com/watch?v=355liCJMWSc&t=5089s

radek8, 2022-08-03:
Yes, there can also be a problem with the SD card, but if the same problem occurs on multiple cards...

radek8, 2022-08-03:
In the video, he also plays with erasing and reloading the bootloader.
But that won't be necessary.
The bootloader is stored elsewhere.

Jannomag, 2022-08-03:
Do you mean the uC named "25Q64JVSIQ"?
If so, it really doesn't look that hard indeed. But it's not easy to get here in Germany without waiting several weeks for China packages.

radek8, 2022-08-03:
I will buy it here for about €3
:-)
https://www.hadex.cz/m518c-pamet-spi-w25q64-64mbit/?gclid=Cj0KCQjwuaiXBhCCARIsAKZLt3mH1lX4Ag7F8e-7H87XYbGAPNUoGFvPLoW05Uvq9EhnritYDVOX2wkaApgXEALw_wcB

Jannomag, 2022-08-03:
I ordered one for 4,50€ on eBay. Also I already ordered a new TFT35 and I just hope I can sell the other one if it works again.

I looked over my small old device collection and checked their pcbs on google but none of them seems to have this chip.

Jannomag, 2022-08-12:
@radek8 I just wanted to let you know that I replaced the chip but it's still not working. It's freezing on updating icons, at different states.

So this doesn't seem to be the problem.

radek8, 2022-08-12:
If you changed the Eeprom and tried multiple SD cards, then I can't think of another reason why this error occurs,

Jannomag, 2022-08-12:
Maybe the processor itself?

Just for double check: the EEPROM is part U6, right?
This one is the chip I replaced

radek8, 2022-08-12:
Yes U6